jgarff/rpi_ws281x #18: PWM1 error

Summer88, 2015-01-02:
Hi jgarff,
on a pi B+ I've tried to change the PIn from PWM0 to PWM1 but get this error for:

LED_PIN     = 19 or 13

~/rpi_ws281x/python/examples $ sudo python strandtest.py
Traceback (most recent call last):
  File "strandtest.py", line 82, in <module>
    strip.begin()
  File "/usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv6l.egg/neopixel.py", line 101, in begin
    raise RuntimeError('ws2811_init failed with code {0}'.format(resp))
RuntimeError: ws2811_init failed with code -1

if I switch PIN on PWM0: LED_PIN     = 18 or 12, everything run fine.
Thanks in advance, BR

jgarff, 2015-01-04:
You can't change the channel simply by changing the pin number.  You must also change the channel being used as well by passing channel=1 to the Adafruit_NeoPixel() class init.

closed 2015-01-04 by jgarff

richteel, 2016-07-03:
Wow, thanks, I am using GPIO 13 on a Raspberry Pi 3 and could not figure out what was wrong. I changed line 81 in strandtest.py by adding , 1 to the Adafruit_NeoPixel and it worked.

strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, 1)