jgarff/rpi_ws281x #20: C++ neopixel API

julesra, 2015-01-23:
Hi, 

I started creating a C++ API for the Neopixel functions (similar to how you provided the Python version of the Neopixel API in /python).

Has this already been done ?

-J
PS: I couldn't figure how to flag this message as a non-issue.

jgarff, 2015-01-24:
Not that I know of.  The low level C interface is very straightforward in my opinion.  I'm not sure it requires a C++ wrapper, as it's already quite easy to incorporate into a C++ project.

julesra, 2015-01-25:
I am familiar with the Neopixel API from working with Arduino (and also from working with the Python API of your library). Now I'm switching to C++, but I'd like to just use my previous code, which uses functions like get / setPixelColor(), show(), etc. 

I'm going to make a C/C++ API for those functions, so that people coming from the Arduino world can use your library more easily.

closed 2015-01-29 by jgarff