jgarff/rpi_ws281x #21: Brightness with setBrightness: 0 brightest and 255 darkest

Thor77, 2015-01-24:
Hi,
if i set the brightness with the setBrightness-function from neopixel.py to 0, the leds are very bright and if i set it to 255, they are very dark. Is this a problem with my LED-Stripe (ws2812b) or did anyone else noticed this and the doc-string should be modified?

jgarff, 2015-01-29:
I haven't seen this the problem.  Is this still a issue for you?

Thor77, 2015-01-29:
Yes: https://github.com/jgarff/rpi_ws281x/pull/22#issuecomment-71337701

jgarff, 2015-02-01:
Unfortunately, I'm still not able to reproduce your issue.  I just tested by adjusting the brightness value in strandtest.py.  At 255 I get full brightness, at 0, it appears off.  Can you please try to replicate using this method?  Thanks.

jgarff, 2015-02-06:
I suspect that your tree is out of date.  There were changes recently which changed the way the brightness works.  Please see:

https://github.com/jgarff/rpi_ws281x/commit/de5a765a57e6c220ae70b570d6631b84163ac841

Prior to this changeset, when using python, there is a side effect of the way that brightness was handled previously.  Please see Tony's comment in:

https://github.com/jgarff/rpi_ws281x/pull/22#issuecomment-71334376

This should no longer be a issue in the latest codebase.

closed 2015-02-06 by jgarff

Thor77, 2015-04-19:
Sorry for my late answer, but i can confirm the issue is solved in the latest codebase, thanks :)