jgarff/rpi_ws281x #23: Modify function-names to fit python-convention and add Python3-Support

Thor77, 2015-01-24:
I think it would be very nice to fit the python-convention in this lib, for example `setPixelColor` should be renamed to `set_pixel_color` and it would be very good to support python3, too.

tdicola, 2015-01-24:
It's a great point to raise, and might be worth spinning off to make another python wrapper around the pixels.  

The reason for the naming in this library is that it matches the names and functionality in the Arduino NeoPixel library (https://github.com/adafruit/Adafruit_NeoPixel).  Although it's not a direct port to go from Arduino code to python, the general idea is that someone familiar with using neopixels on an Arduino can more easily move their code to a Pi using the neopixel.py interface here.  I'd prefer to keep that interface as is, and maybe add a new interface (or even build it as a side library) that's more 'pythonic' and follows the pep8, etc. conventions.

Thor77, 2015-01-24:
Oh, didnt know about this Arduiono-Libary, but thats a really good point.
Maybe i will find some time during the next weeks to create a Python3 'pytonic' version.

jgarff, 2015-02-06:
Please open a new issue or pull request when you have something available.  Thanks.

closed 2015-02-06 by jgarff