jgarff/rpi_ws281x #29: SCons doesn't work on RPi

ismailuddin, 2015-02-28:
This looks like a really cool library, but unfortunately I cannot get it to work on my RPi. I've installed all the necessary dependencies it would require as written in your README file and on the Adafruit website. But when I run scons inside the directory, I just get 'Illegal instruction'. Even tried reinstalled Raspbian on a new SD card, but still same issue... Any ideas?

ismailuddin, 2015-03-01:
After spending the good part of today trying to get this to work, I found the issue is related to using the git command on the RPi itself. For some reason, the files downloaded using git appear corrupted. I cloned this repository using the Git desktop app on Windows, and transferred the files over SFTP to the RPi, and scons worked perfectly fine. Library works brilliantly! The whole git issue though is quite weird...

laboratories, 2015-03-01:
try just downloading the zip of the archive.. that worked fine for me

closed 2015-03-24 by jgarff