jgarff/rpi_ws281x #30: Multiple strings supported?

dokterbob, 2015-03-12:
Hey Jeremy, thanks for this great effort! I've been looking to control NeoPixels straight from a Pi for some time now. 

However, from quickly reading the docs and the code I cannot seem to extract whether or not multiple strings are supported and if so: how. It would be totally awesome if your library could allow something like [Fadecandy](https://github.com/scanlime/fadecandy/) without the additional Teensy.

usefulthink, 2015-03-12:
There are two PWM-Channels on the raspberry-pi that can work independently – so the answer is yes, you can drive two LED-Strips with this library. 

If you look at [main.c](https://github.com/jgarff/rpi_ws281x/blob/master/main.c#L61), there is the initialization for both channels. Both channels work synchronized, so you always need to pass a structure containing both channels to the api-functions (defined in [ws2811.h](https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.h)).

Unfortunately I don't know the correct settings to be used for the second channel (which DMA and GPIO-Pin to use).

tdicola, 2015-03-12:
The pic and table of IO here has the details for the B+ and A+: http://www.raspberry-projects.com/pi/pi-hardware/raspberry-pi-model-b-plus/model-b-plus-io-pins  It's a little tough to read but PWM0 is on GPIO18 and PWM1 is on GPIO13 and GPIO19 (on the far right bottom row in the pic).  Use channel = 1 and pin = 13 or 19 to access the second PWM channel (the first channel 0 is only on pin = 18).

dokterbob, 2015-03-13:
Wow guys, thanks for the quick and extensive reply! If I find the time later today I'll fire a pull req. to put this in the README so other users can profit.

SlySven, 2016-02-25:
Would it be possible to run multiple chains by using addition GPIO pins to control multiplexing of the serial data to multiple strips either at:

* A slow rate: send out two PWM streams to one pair of strips, then switch to a second (or more) pairs - I realise this will also drop the refresh rate by the same factor as the number of multiple strips.

* A fast rate: (and here I may be grasping at straws) by outputting a "low" bit pattern on one PWM and the "high" one on the other and bit-banging between the two of them at a whole bit rate at the points where the levels are the same, using one GPIO pin per strip (I haven't looked at the timing requirements and I am a noobie in this area) - I guess the timing of a switch one per pixel is only a few cycles more and the coding to change the GPIO pin state is not going to be fast enough (which is why we have to used the Hardware PWM to make the signals at the speed anyhow)... :confused:

usefulthink, 2016-02-25:
The first version would indeed be possible, if you just add some AND-Gates between the GPIO-Pin and the LED-Strips. 

Take one of these for example: https://www.nxp.com/documents/data_sheet/74HC_HCT08.pdf

If you take one of these and connect the PWM0 Output from GPIO18 to both Inputs 1A and 2A and two LED-Strips to outputs 1Y and 2Y, you can control which of the two LED-Strips will receive the signal using the Inputs 1B and 2B. Like this:

|  1B  |  2B  | Result      |
|-------|-------|--------------|
|  0    | 0     | none of the strips will receive the signal |
|  0    | 1     | signal will go to second strip (output 2Y)  |
|  1    | 0     | signal will go to first strip (output 1Y)  |
|  1    | 1     | signal is sent to both strips |

The two signal for 1B and 2B can be generated any way you like, so you could use any two GPIO-Pins for this. You just need to do the "channel-select" before calling `ws2811_render()`. I'm note quite sure if that'd be worth the effort though, as you could do pretty much the same by just chaining the two strips together.

Oh, and by the way, in this case you can leave away the logic-level shifter, as the AND-Gates will already do that for you.

SlySven, 2016-02-25:
Yeah, I get you.  The only advantage is if there are times when you want multiple strips to be driven with the same data and others when you don't - or if you have different length strips (different rooms perhaps) and if you could open the individual strips' *gates* just at the right time...

&lt;aside&gt;I am starting to use @joan2937 's [pigpio](https://github.com/joan2937/pigpio) library and was wondering whether it might be possible to [combine the functionality of both](https://github.com/joan2937/pigpio/issues/50) - but as that library use the *Unlicence* to be put in the *Public Domain* the author of that (or anyone else) could not use this and retain that licence - a composite work would have to adopt the BSD-2 line one that is used here...&lt;/aside&gt;

cwolfe007, 2018-05-22:
I am having trouble with this still, trying to run GPIO 13 on PMW channel 1 and GPIO 18 on PMW channel 0. The LEDs only seem to work on GPIO 18 only. The other channels result in wierd flickering. If I code it to use pin 18 only and pin 13 is connected both LEDs start flashing even though I coded it to only go to pin 18. 

Thought I would share and see if anyone else is still having the same issues.

qwertyflagstop, 2018-09-22:
I also experience the flickering with that setup. Any idea why this is happening?

d8ahazard, 2020-12-23:
This is open two years later - can anybody comment on whether or not it's possible to use this to drive multiple strips via multiple GPIO pins at once?

I know mostly we're discussing driving on GPIO 13 and 18, but I'm curious if it is possible to do 4x strips at once, using GPIO 13 and 18, and then MOSI on pin 19 and PCM on pin 40.

Has anybody tried this? What is and isn't working in 2020(1)?

tysonnorris, 2021-02-14:
I logged #446 due to odd results with PWM, not sure if it is related to this issue. But I wanted to add that in my limited tests, I can use GPIO 12 + channel 0 (although with odd results - wrong colors, flickering), but cannot use GPIO 13 + channel 1 (no LEDs light up ever). 

Using GPIO 10 (SPI) worked consistently and correctly, but I too would like a PWM solution if one is possible. 

I cannot use pins 18-21 due to using a sound card that requires those, which admittedly may also be causing problems with PWM I guess.

Xartrick, 2021-02-14:
Audio uses PWM. You cannot drive led strips and use analog audio at the same time.

There is 2 ways : audio over internet or external USB sound-card.