jgarff/rpi_ws281x #33: First Pixel Stuck as Green

jasonsmedley, 2015-04-25:
We are using Richards adapted version of the library and the first pixel in our units keep getting left as green.  Any idea as to what we are doing wrong?

tdicola, 2015-04-27:
Are any other pixels lighting up, or just that first one?  A few really common things to check though are that you have a ground connected between the Pi & pixels.  If there's not a good ground connection and its left floating then you'll usually see randomly lit or even blinking pixels.  Also make sure you're using a level converter (that can handle ~800khz) or diode on the pixel's power supply to make the Pi's 3.3V signal readable by the pixels.  The signal voltage to the pixels has to be above 0.7*Vcc to register as a 1/high signal, so with a 5V power supply the Pi's 3.3V signal is juuust a little too low (0.7*5V = 3.5V).  You can use a level converter to bump the Pi's 3.3V output up to 5V, or you can run your pixel power supply through a big power diode to drop it from 5V down to about 4V and within the range of the Pi's 3.3V output.

archi, 2015-04-29:
jasonsmedley, are you perchance using ws2812 LEDs with an inverted level shifter?
I have that setup and see the same issue, my first pixel always has an additional 100 in the green value.
In my case this seems to be an issue with rpi_ws281x. I suppose those are the two (possible) culprits:
* idle signal not inverted for inverted operation -> ws2812 sees high instead of low when there is no data (not sure where that's handled)
* Color order wrong (there is no ws2812 mode - ws2812 use GRB color order, and the older ws2811 use RGB order -> bad color on 1st pixel, the rest get's aligned by accident/due to bad idle)

Using https://github.com/pimoroni/unicorn-hat/tree/master/python/ws2812 and patching it to run inverted (line 652 in ws2812-RPi.c, use SETBIT instead of CLRBIT on PWM_CTL_POLA1), my setup runs just fine. So this is definitely some issue with rpi_ws281x, I am afraid :(

thejpster, 2015-12-05:
I just fell foul of this issue. archi's patch works for me. Also my 5mm LEDs are WS2812 in GRB format, not RGB.

archi, 2015-12-07:
Ha! Great to hear the patch was useful for someone else but me. Maybe I get around to deploying my Neopixels again, so I could rework the patch to be better suited for inclusion in the master branch.
But for now I don't have time for that :-( Feel free to adapt it yourself and create a PR ;-)

closed 2015-12-31 by jgarff