jgarff/rpi_ws281x #36: Support for rasPi2

iMattmax, 2015-06-17:
Can I use this with the rasPi2 also or is the rasPi2 not supported yet?

usefulthink, 2015-06-22:
There is a fork of this repo that works really well on the rpi2: https://github.com/richardghirst/rpi_ws281x

KuenzelIT, 2015-09-05:
I can confirm that this one does not work as is on my RPI2, but the fork posted by usefulthink works with an unmapping problem when terminating.

jgarff, 2015-10-05:
I just created a rpi2 branch which supports newer kernels and the Raspberry Pi 2.  This is a testing branch, so please give it a shot and let me know if you run into problems.  Thanks

closed 2015-10-05 by jgarff

Vikaton, 2016-01-20:
@jgarff is the rpi2 branch in master now or is it still seperate? I have an rPi2 I'd like to use this library with and I'm wondering which branch I should use.

jgarff, 2016-01-20:
Please use the main branch.  I merged the rpi2 branch into the master a few
weeks ago.

Vikaton, 2016-01-20:
Thanks for the heads up! I'm writing bindings for your library in Rust (https://github.com/Vikaton/ws2811-rs) , would you mind a PR to add a seperate folder for the Rust (like golang) when I'm finished?

jgarff, 2016-01-21:
Sure.  Send me a pull request when you're ready and I'll take a look.