jgarff/rpi_ws281x #37: Only one LED is lit and it does not response properly.

mrminecart, 2015-06-21:
I am using these LEDs: http://www.ebay.co.uk/itm/WS2811-WS2812-5050-RGB-LED-strip-Light-Waterproof-Individually-Addressable-DC5V-/111164910188

I know they work with my Arduino using the FastLED Library.

I am using a SN74HCT08N And chip for my level shifter.

When running ```sudo python strandtest.py``` (or any other application using this library) I get one flashing LED and nothing else.

Video of LED: https://drive.google.com/file/d/0B4ZWPNhO0xpMR3RSUjVFRFVjNlk/view?usp=sharing

mfeif, 2015-06-21:
I've had similar problems. I'm not SURE, but I think that it's a bad LED...
though I've been snipping them off, and find that soon after, the #1 LED
goes bad and starts doing this too.

I'm kindof at my wits' end, really. Wiring these things together ain't
rocket science, and I'm not doing anything weird, but a 1m strand of 30
will happily do its thing for an hour, then start freaking out, then I have
the behavior you're showing.

woverton, 2015-06-21:
These LEDs work perfectly when ran by the arduino library. The issue is certainly with the RPI library not the strips

mfeif, 2015-06-21:
sorry; I'll remove my noisy comment! Thanks.

usefulthink, 2015-06-21:
What kind of power supply are you using? Are all components using the same grounds-reference? (I have seen this kind of behavior when the ground of the power-supply is not connected with the ground of the raspberry and the level shifter).

From software-side it could also be a problem with the drivers for the internal soundcard (see also: https://github.com/raspberry-node/node-rpi-ws281x-native/issues/16). There it would be interesting which operating system you are running and if you are using audio.

jgarff, 2015-12-31:
Closing due to lack of activity.

closed 2015-12-31 by jgarff