jgarff/rpi_ws281x #38: Ubuntu 14.04 - Python 2.7.6 Error with python setup install :(

sirchuck, 2015-07-18:
arm-linux-gnueabihf-gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o -L../. -lws2811 -o build/lib.linux-armv7l-2.7/_rpi_ws281x.so
/usr/bin/ld: .././libws2811.a(ws2811.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
.././libws2811.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

Python 2.7.6 (default, Mar 22 2014, 23:30:12)

Any ideas or solutions ?

kavdev, 2015-09-05:
Happens to me on the latest Ubuntu Make as well.

``pip install -d <somefolder> rpi-ws281x`` (-d downloads without installing)
``cd <somefolder>``
``tar -xvzf rpi_ws281x-1.2.0.tar.gz``
``vim rpi_ws281x-1.2.0/lib/Makefile``
Add ``-fPIC`` to each gcc command as described [here](https://github.com/pimoroni/unicorn-hat/issues/31#issuecomment-127654369).

I'd submit a pull request, but I have no idea how scons works and can't seem to find a Makefile in this repo.

sufzoli, 2015-12-10:
Hi,

I added the following line:

tools_env.Append(CCFLAGS = "-fPIC")

after the

tools_env = clean_envs['userspace'].Clone()

into the SConscript file.
It built successfully with python 3.4 on Ubuntu 14.04
But I still don't know if it is the solution, because didn't have the time to try out the library yet

jgarff, 2015-12-31:
Please try the latest master branch.  I added -fPIC to the compiler flags.  Thanks.

Fixed in commit: 45e4feb5bbeda103b05f984140f33c81e1d06cf8

closed 2015-12-31 by jgarff