jgarff/rpi_ws281x #39: test binary hangs on ws2811.c, line 330

allgoewer, 2015-07-31:
Hello all,

I am currently trying to get the provided test program running.

I'm doing as told by the readme

```
scons
sudo ./test
```

but nothing happens, the process only takes around 100% cpu cycles.

I digged a bit deeper and found it to loop forever in ws2811.c, line 330

```c
while (!(cm_pwm->ctl & CM_PWM_CTL_BUSY))
    ;
```

Surrounding the line with something like
```c
printf("entering loop, file %s, line %d\n", __FILE__, __LINE__); fflush(stdout);
while (!(cm_pwm->ctl & CM_PWM_CTL_BUSY))
    ;
printf("exiting loop, file %s, line %d\n", __FILE__, __LINE__); fflush(stdout);
```

and running it, gives me
```
% sudo ./test
entering loop, file ws2811.c, line 330
```

The funny thing is, the provided python library and `python/example/strandtest.py` work perfectly!

Any idea on how to fix that?

I am using the following software on archlinuxarm, Raspberry Pi B+
```
% uname -r
4.0.9-1-ARCH

% gcc --version
gcc (GCC) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
Dies ist freie Software; die Kopierbedingungen stehen in den Quellen. Es
gibt KEINE Garantie; auch nicht für MARKTGÄNGIGKEIT oder FÜR SPEZIELLE ZWECKE.

% scons --version
SCons by Steven Knight et al.:
	script: v2.3.5.rel_2.3.5:3329:275e75118ad4, 2015/06/20 11:18:26, by bdbaddog on ubuntu1404-32bit
	engine: v2.3.5.rel_2.3.5:3329:275e75118ad4, 2015/06/20 11:18:26, by bdbaddog on ubuntu1404-32bit
	engine path: ['/usr/lib/python2.7/site-packages/SCons']
Copyright (c) 2001 - 2015 The SCons Foundation
```

allgoewer, 2015-08-01:
I nailed it down to gcc.
I reverted back to version 4.9 and it worked again.

Creating assembly output with
```
gcc -c -g -Wa,-a,-ad ws281x.c > ws281x.lst
```
yields

### gcc 4.9
```
297:ws2811.c      ****     // Kill the clock if it was already running
 298:ws2811.c      ****     cm_pwm->ctl = CM_PWM_CTL_PASSWD | CM_PWM_CTL_KILL;
 743              		.loc 2 298 0
 744 0694 10201BE5 		ldr	r2, [fp, #-16]
 745 0698 003092E5 		ldr	r3, [r2]
 746 069c 0030A0E3 		mov	r3, #0
 747 06a0 5A3483E3 		orr	r3, r3, #1509949440
 748 06a4 203083E3 		orr	r3, r3, #32
 749 06a8 003082E5 		str	r3, [r2]
 299:ws2811.c      ****     usleep(10);
 750              		.loc 2 299 0
 751 06ac 0A00A0E3 		mov	r0, #10
 752 06b0 FEFFFFEB 		bl	usleep
 300:ws2811.c      ****     printf("entering loop, %s, %d\n", __FILE__, __LINE__); fflush(stdout);
 753              		.loc 2 300 0
 754 06b4 58009FE5 		ldr	r0, .L42
 755 06b8 58109FE5 		ldr	r1, .L42+4
 756 06bc 4B2FA0E3 		mov	r2, #300
 757 06c0 FEFFFFEB 		bl	printf
 758 06c4 50309FE5 		ldr	r3, .L42+8
 759 06c8 003093E5 		ldr	r3, [r3]
 760 06cc 0300A0E1 		mov	r0, r3
 761 06d0 FEFFFFEB 		bl	fflush
 301:ws2811.c      ****     while (cm_pwm->ctl & CM_PWM_CTL_BUSY)
 762              		.loc 2 301 0
 763 06d4 0000A0E1 		mov	r0, r0	@ nop
 764              	.L41:
 765              		.loc 2 301 0 is_stmt 0 discriminator 1
 766 06d8 10301BE5 		ldr	r3, [fp, #-16]
 767 06dc 003093E5 		ldr	r3, [r3]
 768 06e0 803003E2 		and	r3, r3, #128
 769 06e4 000053E3 		cmp	r3, #0
 770 06e8 FAFFFF1A 		bne	.L41
 302:ws2811.c      ****         ;
 303:ws2811.c      ****     printf("exiting loop, %s, %d\n", __FILE__, __LINE__); fflush(stdout);
 771              		.loc 2 303 0 is_stmt 1
 772 06ec 2C009FE5 		ldr	r0, .L42+12
 773 06f0 20109FE5 		ldr	r1, .L42+4
 774 06f4 28209FE5 		ldr	r2, .L42+16
 775 06f8 FEFFFFEB 		bl	printf
 776 06fc 18309FE5 		ldr	r3, .L42+8
 777 0700 003093E5 		ldr	r3, [r3]
 778 0704 0300A0E1 		mov	r0, r3
 779 0708 FEFFFFEB 		bl	fflush
```

### gcc 5.1
```
297:ws2811.c      ****     // Kill the clock if it was already running
 298:ws2811.c      ****     cm_pwm->ctl = CM_PWM_CTL_PASSWD | CM_PWM_CTL_KILL;
 848              		.loc 2 298 0
 849 0834 10301BE5 		ldr	r3, [fp, #-16]
 850 0838 0020D3E5 		ldrb	r2, [r3]
ARM GAS  /tmp/ccf6W8AS.s 			page 23

 851 083c 0020A0E3 		mov	r2, #0
 852 0840 202082E3 		orr	r2, r2, #32
 853 0844 0020C3E5 		strb	r2, [r3]
 854 0848 0120D3E5 		ldrb	r2, [r3, #1]
 855 084c 0020A0E3 		mov	r2, #0
 856 0850 0120C3E5 		strb	r2, [r3, #1]
 857 0854 0220D3E5 		ldrb	r2, [r3, #2]
 858 0858 0020A0E3 		mov	r2, #0
 859 085c 0220C3E5 		strb	r2, [r3, #2]
 860 0860 0320D3E5 		ldrb	r2, [r3, #3]
 861 0864 0020A0E3 		mov	r2, #0
 862 0868 5A2082E3 		orr	r2, r2, #90
 863 086c 0320C3E5 		strb	r2, [r3, #3]
 299:ws2811.c      ****     usleep(10);
 864              		.loc 2 299 0
 865 0870 0A00A0E3 		mov	r0, #10
 866 0874 FEFFFFEB 		bl	usleep
 300:ws2811.c      ****     printf("entering loop, %s, %d\n", __FILE__, __LINE__); fflush(stdout);
 867              		.loc 2 300 0
 868 0878 4B2FA0E3 		mov	r2, #300
 869 087c 8C109FE5 		ldr	r1, .L44
 870 0880 8C009FE5 		ldr	r0, .L44+4
 871 0884 FEFFFFEB 		bl	printf
 872 0888 88309FE5 		ldr	r3, .L44+8
 873 088c 003093E5 		ldr	r3, [r3]
 874 0890 0300A0E1 		mov	r0, r3
 875 0894 FEFFFFEB 		bl	fflush
 301:ws2811.c      ****     while (cm_pwm->ctl & CM_PWM_CTL_BUSY)
 876              		.loc 2 301 0
 877 0898 0000A0E1 		mov	r0, r0	@ nop
 878              	.L43:
 879              		.loc 2 301 0 is_stmt 0 discriminator 1
 880 089c 10301BE5 		ldr	r3, [fp, #-16]
 881 08a0 0020D3E5 		ldrb	r2, [r3]
 882 08a4 7220EFE6 		uxtb	r2, r2
 883 08a8 0110D3E5 		ldrb	r1, [r3, #1]
 884 08ac 7110EFE6 		uxtb	r1, r1
 885 08b0 0114A0E1 		mov	r1, r1, asl #8
 886 08b4 022081E1 		orr	r2, r1, r2
 887 08b8 0210D3E5 		ldrb	r1, [r3, #2]
 888 08bc 7110EFE6 		uxtb	r1, r1
 889 08c0 0118A0E1 		mov	r1, r1, asl #16
 890 08c4 022081E1 		orr	r2, r1, r2
 891 08c8 0330D3E5 		ldrb	r3, [r3, #3]
 892 08cc 7330EFE6 		uxtb	r3, r3
 893 08d0 033CA0E1 		mov	r3, r3, asl #24
 894 08d4 023083E1 		orr	r3, r3, r2
 895 08d8 803003E2 		and	r3, r3, #128
 896 08dc 000053E3 		cmp	r3, #0
 897 08e0 EDFFFF1A 		bne	.L43
 302:ws2811.c      ****         ;
 303:ws2811.c      ****     printf("exiting loop, %s, %d\n", __FILE__, __LINE__); fflush(stdout);
 898              		.loc 2 303 0 is_stmt 1
 899 08e4 30209FE5 		ldr	r2, .L44+12
 900 08e8 20109FE5 		ldr	r1, .L44
 901 08ec 2C009FE5 		ldr	r0, .L44+16
 902 08f0 FEFFFFEB 		bl	printf
ARM GAS  /tmp/ccf6W8AS.s 			page 24

 903 08f4 1C309FE5 		ldr	r3, .L44+8
 904 08f8 003093E5 		ldr	r3, [r3]
 905 08fc 0300A0E1 		mov	r0, r3
 906 0900 FEFFFFEB 		bl	fflush 
```

allgoewer, 2015-08-03:
Fixed.
See allgoewer/rpi_ws281x@46c8a4c1218ce999bb6371b2a20edce2c6a0bfee

JIP0815, 2015-08-03:
Hi,

what do you think about adding this function:

def getPixelColorRGB(self, n):
     """Get the RGB values for the LED at position n."""
     RGBint = self._led_data[n]
     Blue =  RGBint & 255
     Green = (RGBint >> 8) & 255
     Red =   (RGBint >> 16) & 255

     return (Red, Blue, Green)

This function makes it possible to read out the RGB color value of the 
single pixels.

Greetings,
jip0815

jgarff, 2015-12-31:
Thanks for tracking this down.  That's very interesting.  I don't believe however that taking the packed attribute off of the structure is the right approach to fix this problem.  You're right, in that everything is already 32-bit aligned, so it shouldn't matter, but having packed hardware structures is always the right thing to do.  Instead of working around this compiler bug by removing the packed attribute, I think it would be better to make some macros or static inlines and perform reads/writes to the registers though them instead.

This isn't the first time I've seen compilers do this in the optimization path, it's somewhat common.  This is why many kernels have readl()/writel() helpers for register manipulation.  I'll put this on my list of things to do, but if you're willing to make the change, I'm happy to consider a pull request instead.  Thanks.

jgarff, 2015-12-31:
jip0815:  Thanks.  I've actually created and used a similar function in some of my personal projects.  If you'll submit a pull request with the change, I'll include it.

closed 2015-12-31 by jgarff