jgarff/rpi_ws281x #40: Hangs on strandtest.py

bbbenji, 2015-08-22:
I am running rpi_ws281x on RPi B ArchArm with a NeoPixel Ring 24. After starting strandtest.py nothing happens. I am unable to terminate and no output is visible. How can I start to diagnose this?

KuenzelIT, 2015-09-05:
I got the same problem. It works with this fork, while producing a unmap error at the end:
https://github.com/richardghirst/rpi_ws281x

antti004, 2015-10-13:
Same thing. Hangs at strip.begin()
Using RPI2.

electrodynatronic, 2015-11-09:
Another here with the same problem. This lib works fine with the single core rPi, but does not work with rPi 2.

usefulthink, 2015-11-10:
@antti004 @electrodynatronic Are you using the new [rpi2-branch](https://github.com/jgarff/rpi_ws281x/tree/rpi2)? The version in master doesn't support the rpi2.

electrodynatronic, 2015-11-10:
@usefulthink Nay sir, I had failed to notice that! TYVM.

m4rkireland, 2015-12-08:
@electrodynatronic is it working for you now? I am using the rpi2-branch and still having issues. ./test runs fine.

jgarff, 2015-12-31:
I've now merged the rpi2 branch into the master.  Please re-test and re-open if necessary.  Thanks.

closed 2015-12-31 by jgarff