jgarff/rpi_ws281x #41: "Can't open device file /tmp/mailbox-XXXX: No such device or address" - Kernel 4.1.6

Gadgetoid, 2015-08-24:
It looks like Kernel 4.1.6 has introduced some changes which have broken the use of Mailbox functionality in this library. I've updated to the latest Raspbian, `Linux 4.1.6-v7+ #810` and am no longer able to get it working.

The error seems to be with opening the device after creation, despite creation seemingly happening fine and not throwing any errors:

`Can't open device file /tmp/mailbox-XXXX: No such device or address`

I've seen at least one other report of this problem in respect to our Unicorn HAT library, and another seemingly related problem on this thread here: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=117547

I've tried stabbing in the dark to see if I can get it working again- but I'm a little out of my depth!

Gadgetoid, 2015-08-24:
Turns out the fix for this is to prefer `/dev/vcio` and fall back to creating a mailbox device when it doesn't exist. I've tested this on Pi 2, Kernel 4.1.6 and it appears to work. Relevant code here: https://github.com/pimoroni/unicorn-hat/commit/5c03f81b72eb2e44c58f850db4e3fad90dc3d7a7

tejonbiker, 2015-08-25:
Thanks you very much, it's now working :D

limpygnome, 2015-08-25:
Thank you so much, working here on the Pi 2 on 4.1.6 too!

Is /dev/vcio used on the Pi for memory mapping access to the GPIO?

usefulthink, 2015-08-26:
@limpygnome iirc, the memory-mapped I/O related to DMA and PWM-Setup (you could call that GPIO-handling) is done using raw physical memory (`/dev/mem`). 

`/dev/vcio` (or the previous version using `/tmp/mailbox-xxxx` devices) is a feature provided by the video-core, where the driver borrows memory to store the PWM-data to be transferred to the PWM-module using DMA.

(this is how I understood and researched it so far, please correct me if I got something wrong here)

jgarff, 2015-10-05:
I just created a rpi2 branch which supports newer kernels and the Raspberry Pi 2.  This is a testing branch, so please give it a shot and let me know if you run into problems.  Thanks.

closed 2015-10-05 by jgarff

tsuckow, 2015-11-07:
On an rpi it never seems to get past init.

jgarff, 2015-11-07:
Did you sync the rpi2 branch, and is this happinging on a pi2?

tsuckow, 2015-11-08:
RPI 1B running Arch linux arm pi (alarmpi)

The main branch doesn't work anymore because of the mailbox issue. The rpi2 branch gets past that but hangs on the CM_PWM_CTL_BUSY tight loop.

jgarff, 2015-11-08:
Please make sure your audio device isn't being used.  The raspberry pi uses the PWM controller for the analog audio audio, so it's not possible to use both at the same time.

wefalk, 2015-11-24:
Still get error:  "Can't open device file /tmp/mailbox-10709: No such device or address" on rpi2 branch

System:  Linux raspberrypi 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux

hjoertel, 2016-01-10:
I stumbled here as well when trying to control ws2812 LED stripes using the PiZero. PiZero came with kernel 4.1.13.
I did only change mailbox.h definitions for the vcio driver major number and the ioctl code.

keerthanamanivannan, 2016-07-20:
Hey hjoertel, same here. I'm trying to run this library on RPi Zero with neopixel ws2812 ring. I'm getting this error:
Can't open device file /tmp/mailbox-12666: No such device or address
Can't open device file /tmp/mailbox-12666: No such device or address
Can't open device file /tmp/mailbox-12666: No such device or address

What did you change the vcio driver major number to? And what's this change in ioctl code?

wefalk, 2016-07-20:
Hi hjortel:

What I did was to enable the use of the device tree from within raspi-config.  That fixed it!  No other changes.

Hope it works for you, please let me know either way…

wefalk

keerthanamanivannan, 2016-07-20:
Hi wefalk,
How'd you enable device trees from raspi-config? I go to Advanced options and there's no device tree option..

keerthanamanivannan, 2016-07-20:
And did you also get the /tmp/mailbox error? How did you fix it?

wefalk, 2016-07-20:
I was getting the error until turning on the device tree, then the error disappeared…

And did you also get the /tmp/mailbox error? How did you fix it?

keerthanamanivannan, 2016-07-20:
Wait, how'd you turn on the device tree?

wefalk, 2016-07-20:
Perhaps it does not exist on the Pi Zero, it did work on the Pi 2.  I have no experience with the Zero at this point.

Best of luck

wefalk

keerthanamanivannan, 2016-07-20:
oh ok thanks. I'm just poking around here and there trying to fix the mailbox error. Hoping @hjoertel or @jgarff will give me answers to fix it!

wefalk, 2016-07-20:
Good luck!

hjoertel, 2016-07-20:
Am Mittwoch, 20. Juli 2016, 12:52:10 schrieb Keerthana Subramanian Manivannan:
> oh ok thanks. I'm just poking around here and there trying to fix the mailbox error. Hoping @hjoertel or @jgarff will give me answers to fix it!

Hi Keerthana,

sorry for letting this topic open. It's working for month now and I don't know what I did to get it working.
Unfortunately I'm just not able to look at it. If you can wait 7 to 8 days? When I#m back at my PC I'll look what i will find.
Please check http://oertel-halle.de/projects/LEDstripes where I wrote down something I did.

regards
   Heinz

-- 
mit freundlichen Grüßen
       Heinz-Jürgen Oertel

jgarff, 2016-07-24:
Which distro are you running?  It seems that it's lacking some Broadcom kernel support which is required to get to the right type of non-cacheable memory.  I know that the default raspbian image supports it, you might try running that instead.  You also need to remember to run as root.