jgarff/rpi_ws281x #45: Gamma correction

russau, 2015-09-14:
Would you consider adding [gamma correction](https://learn.adafruit.com/led-tricks-gamma-correction/the-issue) to this library?

I'm new to this sort of thing, and it took to a while to find the source of my "colours look wrong" issue.

piers7, 2015-10-14:
@jgarff The Pimoroni version of this library already includes gamma correction, and I've already merged it my fork of your code, so if you are interested I could easily push a PR

russau, 2015-10-14:
I'm defn interested in seeing the PR

archi, 2015-10-14:
The downside with gamma correction is the loss of different color values, as outlined in the link.
With their gamma correction of 2.8 the 256 colors become just 163.
Has anyone tried temporal dithering to produce more colors?
I believe this should also be added in case gamma correction is used widespread...

In case you want gamma correction just for a few projects, they show the code on the linked website.
All you need to do is write a mapping function and apply it when sending the color value to the library - which you might be doing in a single place, anyway ;-)

    inline int mapGamma (int input) {
        //pseude code, but should work
        int max_in = 255;
        int max_out = 255;
        float gamma = 2.8;
        return (int)pow((float)input / (float)max_in, gamma) * max_out + 0.5;
    }

piers7, 2015-10-14:
So the Pimoroni fork just takes the lookup table approach. Check out:
https://github.com/pimoroni/unicorn-hat/blob/master/python/rpi-ws281x/lib/gamma.h
which just gets applied in ws2811_render (search for 'gamma')
https://github.com/pimoroni/unicorn-hat/blob/master/python/rpi-ws281x/lib/ws2811.c

archi, 2015-10-14:
As I said, for the RPi there is no reason not to do the float math (or filling the 1D gamma/color lookup table once during startup, in case I am underestimating the complexity of a pow() on ARM).

But that's not my real concern:

The Pimoroni fork only produces 163 colors per channel, for a total of 163^3 = 4330747 colors on the LED. With the 256 colors per channel, you end up at 16777216 colors.
So you gained gamma correction and orange looks orange, but: You just lost about 74% of your colors.

And some people are scaling down the values to reduce brightness -> they are already reducing their color space.

Just because the unicorn-hat people are doing it, this is not necessarily the right thing to do for everyone. (Even though gamma correction is a great idea!).

But I am not the maintainer (my PR for something else is waiting since ever), these are just my 2c on it ;-)

piers7, 2015-10-14:
Fair comment about the cost of the math, but in terms of the color loss I'm just inclined to shrug. What's the benefit of having extra color points if they're *not the right colors*? 4M different shades is still quite a few IMHO. And if the gamma correction isn't in the library it just increases the complexity of use for all clients.
Perhaps a good compromise would be to have the gamma correction optional in the library (flag on startup) so if the caller wants to use a more sophisticated scheme instead (eg temporal dithering) they can.

jgarff, 2015-12-31:
@piers7 Please send a pull request and I'll take a look.  You probably already saw this, but I believe there is already a mechanism to perform correction in the python library, but not in C.  Seems like it would be useful in C as well.  Thanks.

closed 2015-12-31 by jgarff