jgarff/rpi_ws281x #46: Onion Omega

ballyrick, 2015-11-18:
Trying to get this working on the Onion (using python-light) and I am getting a lot of python errors when running setup.py

Traceback (most recent call last):
  File "setup.py", line 3, in <module>
    from ez_setup import use_setuptools
  File "/root/rpi_ws281x-master/python/ez_setup.py", line 19, in <module>
    import tempfile
  File "/usr/lib/python2.7/tempfile.py", line 35, in <module>
    from random import Random as _Random
  File "/usr/lib/python2.7/random.py", line 49, in <module>
    import hashlib as _hashlib
  File "/usr/lib/python2.7/hashlib.py", line 149, in <module>
    import logging
ImportError: No module named logging

Any ideas what is happening?
P.S Great library!

jgarff, 2015-11-18:
This error indicates that your python/setuptools installation is missing
the logging module.  This might be as simple as missing a python package in
your distro (though this would surprise me as it should have been installed
as a dependency).  I'd double check to make sure you have the
python-logbook package installed.

Jeremy

ballyrick, 2015-11-18:
Hi,

Thanks for the very fast reply.
python-logbook isn't available to install but i installed python-logging.
After rerunning the setup.py i got a crazy amount of errors (http://pastebin.com/VEaDNT1R)

Is there anyway to compile the python on my PC and copy it over to the onion?

usefulthink, 2015-11-18:
@ballyrick I wouldn't be too confident that this will work out unless you are planning on rewriting significant parts of the library (most of which is pretty specific to the Broadcom-Chips (BCM2835/2836) found in the Raspberry PIs). 

The hardware of the omega (Atheros AR9331) will very likely behave completely different (and, as this library will at least try to access certain regions of physical memory, it might even harm your device. So be sure you know what you are doing).

A quick googling did not turn up anything useful considering driving neopixels/ws2812 LEDs from the onion.

ballyrick, 2015-11-18:
Thanks @usefulthink .
I never looked at it that way, I just got the onion from kick starter and saw it used python so tried using the nano pixel with it. 

Thanks for the help guys, hopefully someone ports the code in the future.

What I might do for the time being is control my nanoPixel with arduino and use the onion to control the arduino.

jgarff, 2015-11-19:
Sorry, I didn't know what the onion was in my previous reply.
Unfortunately as mentioned by others, this library only works on the pi,
and is Broadcom hardware specific.

Jeremy

closed 2015-11-19 by jgarff