jgarff/rpi_ws281x #47: Incompatible with pyaudio?

skiniry, 2015-11-28:
So I've encountered a strange issue with my lights while using the pyaudio module. Using this module in any script on the pi seems to cause the lights to behave erratically. Running a script with just:

import pyaudio
p = pyaudio.PyAudio()

and then running the strandtest.py will cause the lights to blink rapidly on and off in a series of random colours. This persists until the pi is rebooted.

jgarff, 2015-11-28:
That's correct.  The Pi only has a single PWM hardware module.  The audio
device uses the PWM, as does this library.  If you need audio, you'll
probably want to use a USB audio device instead.

skiniry, 2015-11-28:
Will do, Thanks.

closed 2015-11-28 by skiniry

doudz, 2015-12-17:
I know this issue is closed but is there a way to use PCM peripheral instead of PWM to avoid audio conflict ?
As I understand ServoBlaster does it so controlling servo is possible with audio output.
https://github.com/richardghirst/PiBits/tree/master/ServoBlaster