jgarff/rpi_ws281x #50: Use PCM instead of PWM to avoid audio conflict

doudz, 2015-12-26:
Is there a way to use PCM peripheral instead of PWM to avoid audio conflict ?
As I understand ServoBlaster does it so controlling servo is possible with audio output.
https://github.com/richardghirst/PiBits/tree/master/ServoBlaster

(Maybe this is not possible, I don't know how this really work)

jgarff, 2015-12-31:
It might be possible.  I haven't done too much research into it.  If I remember correctly though, the project you're referring to uses the PCM hardware to generate interrupts, which the driver then uses to modify the GPIO states in software.  Because of this, there is a high amount of jitter in the signal and the frequency is lower by a order of magnitude.  This works fine for servos, which are much lower frequency, but not for WS2812s.  Instead it might be possible to use the PCM hardware directly, without software modifying the GPIOs, but this would mean that you could only use the PCM DOUT pin to drive the LEDs.  There might also be restrictions of the format of the DOUT signal, but I haven't looked closely enough to see.

closed 2015-12-31 by jgarff

sufzoli, 2016-01-11:
I'm just asking: This problem connected to both of the PWM channels or there is a chance to use PWM1 for the LEDs and keep the audio on the PWM0?

jgarff, 2016-01-11:
Unfortunately, no.  The Broadcom chip doesn't have two discrete pieces of PWM hardware.  It a single controller with a single set of control registers, interpreting a single interleaved data stream, with two output channels.  Besides, I believe the audio uses both channels anyway.  One for left and the other for right speakers.  Nonetheless, it's one piece of hardware and registers that the drivers don't know how to share.