jgarff/rpi_ws281x #54: RunTime error: ws2811_init failed on raspberry pi A+ using Raspian

InterstellarSkies, 2016-01-08:
Hey I was wondering if there is any support for neopixels on raspberry pi A+? After running the python code example strandtest.py, I keep getting runtime error: ws2811_init failed

Thanks in advance!

jgarff, 2016-01-08:
The A+ should be supported.  Please make sure you're running as root, and you're not using audio at the same time.

closed 2016-01-08 by jgarff

InterstellarSkies, 2016-01-09:
I just don't know where anything is going wrong, I strictly followed the adafruit raspberry pi neopixel tutorial and after executing 
`sudo python strandtest.py`

I still get 

```
Traceback(most recent call last):
File "strandtest.py" , line 83, in <module>
strip.begin()
File "build/bdist.linux-armv61/egg/neopixel.py", line 100, in begin
RuntimeError: ws2811_init failed with code -1
```
and there is nothing else running on the raspberry pi

heschlie, 2016-04-05:
Were you ever able to resolve this?  I am running into the same error on the RPi 3 and seem to have hit a wall.

SlySven, 2016-04-05:
That is a failure in `ws2811.c` in the `int ws2811_init(ws2811_t *ws2811)` function.  I wonder whether the code that checks the hardware and tweaks setting according to the model/revision is right for your case(s) (I am assuming there is such and it is needed for this project!)

@heschlie you will need to be using a version fresh enough to know about the RPi 3B...!

\**an innocent passing by-stander with a possible future use of* rpi_ws281x \*

heschlie, 2016-04-05:
@slysven yea I think you're right, was doing some mulling around and saw the additions in upstream, going to attempt to merge those in and build it tonight.