jgarff/rpi_ws281x #55: Doesnt work ?

jonshouse1, 2016-01-09:
root@raspberrypi:~/rpi_ws281x-master# ./test
Can't open device file
: Permission denied

usefulthink, 2016-01-09:
@jonshouse1 I'd love to help, but we need some more information in order to know what is going wrong with your setup. 

 * Which raspberry-pi model are you using? 
 * Are you running the latest Version of this library from the master-branch? 
 * Which distribution and version is running on your raspberry (for instance raspbian/jessie)?

jonshouse1, 2016-01-09:
Model is a B

Library was downloaded today, so I assume it is the latest
root@raspberrypi:~# md5sum rpi_ws281x-master.zip 
be8515cce9714d2bd608e36ad837d7d3  rpi_ws281x-master.zip

root@raspberrypi:~# uname -a
Linux raspberrypi 3.10.25+ #622 PREEMPT Fri Jan 3 18:41:00 GMT 2014 armv6l GNU/Linux

Distro is Raspian, probably not the most up to date but to avoid some the recent bugs I prefer to stick to the version that is reliable for my embedded work if possible.

root@raspberrypi:~# cat /etc/debian_version 
7.2

root@raspberrypi:~# /opt/vc/bin/vcgencmd version
Jan  6 2014 21:19:57 
Copyright (c) 2012 Broadcom
version b00bb3ae73bd2799df0e938b7a5f17f45303fb53 (clean) (release)

Thanks for the help,
Jon

InterstellarSkies, 2016-01-15:
@jonshouse1 I didn't quite have the same problem however I found using an older commit worked, I suggest https://github.com/jgarff/rpi_ws281x/tree/de5a765a57e6c220ae70b570d6631b84163ac841 hopefully that may help you.

As for addressing the problem I cannot help you, sorry!

jonshouse1, 2016-01-15:
That version builds and works, many thanks :-)

rodrigob, 2016-02-18:
Just tried the latest git version on a rpi2, and it worked out of the box.

rodrigob, 2016-02-25:
Did you try "sudo ./test" ?

markwal, 2016-03-02:
I've noticed that it works on an rpi2 on jessie-lite, but not on the latest images from retropie or osmc.  There is something about those images (a driver, config or something) that causes the timing to be wrong.  I hooked it up to my oscilloscope and the high time seems about right (400ns), but the low time is wrong (~5000ns).  There are very short portions of the signal that look correct, but most of the bits are >10x too slow (75kbps).  I haven't figured out what it is about the images that interfere with rpi_ws281x yet.

jgarff, 2016-03-10:
It sounds like something else is also using the PWM module, and interfering with the library.  Please make sure your audio is disabled.  The built-in audio on the Pi uses the PWM module and doesn't work with this library.  If you need audio along with LED control, you'll need to use a USB audio device instead.  I've successfully done this with my projects.

closed 2016-03-10 by jgarff