jgarff/rpi_ws281x #56: On GRB and RGB

Vikaton, 2016-01-21:
In the guide here (https://learn.adafruit.com/neopixels-on-raspberry-pi/wiring) it states:
```
Also be aware the rpi_ws281x library only supports GRB NeoPixels/WS281x LEDs right now.  Support for RGB and BGR will likely come in the future (but pull requests are happily accepted!).
```

and I was looking to buy https://www.adafruit.com/products/1586, but according to the guide, This wouldnt work because the LED ring is RGB? does this library now support RGB? (the guide hasn't been updated lately) (I personally never heard of GRB or BGR before, but I thought to ask before buying anything)

jgarff, 2016-01-21:
I believe that document is a bit out of date.  The latest code supports
other color variations.

Vikaton, 2016-01-21:
Okay, I'm going to order and try it out, Ill close this for now.

closed 2016-01-21 by Vikaton