jgarff/rpi_ws281x #59: RGBW support

abergland, 2016-02-08:
Hi,

I'm trying to control a RGBW NeoPixel strip (https://www.adafruit.com/products/2847) with a Raspberry Pi 2 (model B, 1Gb RAM).

I've been using the rpi2 branch and it is able to control the LEDs. But, as you would probably expect, the behavior of the LEDs is not correct. This clearly has something to do with how the R, G, B, and W information is encoded. For instance, if I try to turn on an odd numbered LED (say LED 5 in my strip), the blue light does not turn on and the white light is turned on on LED #4; if I try to turn on an even numbered LED, the blue & white LED turn on simultaneously.

I've tried mucking around in neopixel.py and ws2811.c but I cannot seem to sort it out (I'm not terribly proficient at byte-code type stuff). From ws2811.c, it would seem that the appropriate changes would be around lines 58-62 where the LED_BIT_COUNT and PWM_BYTE_COUNT variables are set as well as around lines 636 where the 'r/g/bshift' variables are set. I tried just specifying them as +24, +16, +8, +0, but that didn't seem to do much.

Can anyone point me in the right direction about how to get RGBW control using this library or incorporate a fix? 

Also, here is the data sheet for the lights which describes the LEDs (https://www.adafruit.com/images/product-files/2757/p2757_SK6812RGBW_REV01.pdf)

Thanks!
Alan

usefulthink, 2016-02-08:
in the `ws2811_render`-function you also need to write out the values for the white-channel. (i.e. add the white-value to the `color`-array, see here: https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.c#L638-L645)).
The other thing would be to adjust the `LED_BIT_COUNT` from 3 to 4 byte per pixel. The memory-management functions etc should all rely on that.

I would love to see how good this works, never seen these new LEDs before...

abergland, 2016-02-08:
So, I've tried a few things along these lines:

I changed the LED_BIT_COUNT to `((leds * 4 * 8 * 3) + ((LED_RESET_uS * (freq * 3) / ...`

I also changed  `ws2811_render` to:

`int rshift = (channel ->strip_type >> 24) & 0xff`
`int gshift = (channel ->strip_type >> 16) & 0xff`
`int bshift = (channel ->strip_type >> 8) & 0xff`
`int wshift = (channel ->strip_type >> 0) & 0xff`

and also added `wshift` to the color array.

Finally, I wrote a new function in the neopixels.py file to encode the RGBW into a 32bit.

`def ColorRGBW(red, green, blue, white):`
`     return((red<<24) | (green<<16) | (blue<<8) | white`

My suspicion is that something in the part that pads to 32 bit needs to be modified 
https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.c#L62-L63

I've tried modifying it in various ways but I don't understand the code well enough to modify it in an intelligent manner (I've just been trying to break it, but can't seem to).

rodrigob, 2016-02-11:
I just ordered a RGBW strip too (and the 3.3v->5v logic level shifter).
@abergland how did things go ?

rodrigob, 2016-02-18:
Just got my neopixels. Tried the RGB ones (on rpi2), and the library works without any problem.
Tomorrow I will give it a try at the RGBW ones.

rodrigob, 2016-02-24:
From  a quick test, after doing the suggested changes things compile, but out of the box I get R, G, and W colours working, but B (blue) is not activating. Now searching for the culprit...

rodrigob, 2016-02-24:
More clues:
a) Noticed that channel->strip_type needs to be set to the right value. Otherwise it the wshift will be wrong. I am currently trying with  `WS2811_STRIP_WRGB  0x18100800`
b) On a closer inspection, it is both blue _and_ white that activate on the stripe.

So the actual problem is:
R, G, B work fine; but W triggers with B, and when trying to set W alone everything stays dark.

It is a if "the blue value was sent twice". More investigation needed. 
Any suggestions of what to try is welcome.

rodrigob, 2016-02-24:
Turns out I was just doing dumb compilation error (`python setup.py install` does check for changes in the linked libraries). After changing the `channel->strip_type` to `WS2811_STRIP_WRGB 0x18100800` then I can properly control the colours of strip.

Yey !

rodrigob, 2016-02-24:
Pushed my changes into
https://github.com/rodrigob/rpi_ws281x/commit/c214517716753af4203d7e0eb2b32429856b555f

right now there is not a "switch" to pass from RGBW to RGB, or to support RGBW in one channel and RGB in the other.  

That would be the proper way to close this issue.

abergland, 2016-02-26:
Awesome. Thanks! Works great.

closed 2016-02-26 by abergland