jgarff/rpi_ws281x #60: Not working correctly on Raspberry Pi 2

richardxyx, 2016-02-08:
I'm trying to get my WS2812 5050 LED Strip (5m, 150 LEDs) to work on my Raspberry Pi 2. I've tried several different libraries and while some don't start at all, others (including this one) make about half of my LED Strip flicker in random colors and without any order when checking it with the strandtest.py.

After I read some issues and pull requests, it seems like I can't find a solution for my problem. I also tried messing with the frequency and used Pin 12 instead of Pin 18 without any improvement.

If it helps, here is my install routine:

```Shell
pi@pi:~ $ sudo rm -r rpi_ws281x/
pi@pi:~ $ git clone https://github.com/jgarff/rpi_ws281x.git
Cloning into 'rpi_ws281x'...
remote: Counting objects: 216, done.
remote: Total 216 (delta 0), reused 0 (delta 0), pack-reused 216
Receiving objects: 100% (216/216), 141.10 KiB | 0 bytes/s, done.
Resolving deltas: 100% (125/125), done.
Checking connectivity... done.
pi@pi:~ $ cd rpi_ws281x/
pi@pi:~/rpi_ws281x $ sudo scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
CC      main.o
CC      mailbox.o
CC      ws2811.o
CC      pwm.o
CC      dma.o
CC      rpihw.o
AR      libws2811.a
RANLIB  libws2811.a
LINK    test
scons: done building targets.
pi@pi:~/rpi_ws281x $ cd python/
pi@pi:~/rpi_ws281x/python $ sudo python setup.py install
running install
running bdist_egg
running egg_info
creating rpi_ws281x.egg-info
writing rpi_ws281x.egg-info/PKG-INFO
writing top-level names to rpi_ws281x.egg-info/top_level.txt
writing dependency_links to rpi_ws281x.egg-info/dependency_links.txt
writing manifest file 'rpi_ws281x.egg-info/SOURCES.txt'
reading manifest file 'rpi_ws281x.egg-info/SOURCES.txt'
writing manifest file 'rpi_ws281x.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
copying neopixel.py -> build/lib.linux-armv7l-2.7
running build_ext
building '_rpi_ws281x' extension
swigging rpi_ws281x.i to rpi_ws281x_wrap.c
swig -python -o rpi_ws281x_wrap.c rpi_ws281x.i
creating build/temp.linux-armv7l-2.7
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o -L../. -lws2811 -o build/lib.linux-armv7l-2.7/_rpi_ws281x.so
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
copying build/lib.linux-armv7l-2.7/_rpi_ws281x.so -> build/bdist.linux-armv7l/egg
copying build/lib.linux-armv7l-2.7/neopixel.py -> build/bdist.linux-armv7l/egg
byte-compiling build/bdist.linux-armv7l/egg/neopixel.py to neopixel.pyc
creating stub loader for _rpi_ws281x.so
byte-compiling build/bdist.linux-armv7l/egg/_rpi_ws281x.py to _rpi_ws281x.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying rpi_ws281x.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying rpi_ws281x.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying rpi_ws281x.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying rpi_ws281x.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
writing build/bdist.linux-armv7l/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg
Removing /usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg
Copying rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg to /usr/local/lib/python2.7/dist-packages
rpi-ws281x 1.0.0 is already the active version in easy-install.pth

Installed /usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg
Processing dependencies for rpi-ws281x==1.0.0
Finished processing dependencies for rpi-ws281x==1.0.0
pi@pi:~/rpi_ws281x/python $ sudo python examples/strandtest.py 
Press Ctrl-C to quit.
```

richardxyx, 2016-02-08:
Alright. After more than 12h of debugging, I finally found my small little flaw in the system. I forgot to ground the Pi with the ground of the LED strip. A simple jumper wire between both grounds solved the problem. Still, only about half of the strip light up, but that may be a problem with the strip itself.

closed 2016-02-08 by richardxyx

jgarff, 2016-02-08:
> I'm trying to get my WS2812B 5050 LED Strip (5m, 150 LEDs) to work on my

This sounds suspiciously like a issue with your hardware.  I've seen this
behavior when a faulty level translator is used.  I'd double check to make
sure your level translator is working properly and that your LEDs are
properly grounded and powered.  If you have access to a oscilloscope, you
could verify that proper signaling is being received/sent.