jgarff/rpi_ws281x #63: User Selectable RGB order flipped on render build

dmjpixel, 2016-02-24:
Came across this testing RPI_WS281X/GPIO connection on Pi2.  A RGB pixel flood I built, and works as RGB on other controllers, was displaying as GBR via the FPP test page.

The proposed pull request #62 changes the render order of build to match the order of the bit shift routine.  The bit shift routine "forces" an arrangement to a RGB order based
on color order selected by user but the render build actually loads it in GRB
order. This change will render as the bits were populated based on user selection.

Tested on a known RGB built 2811 flood on Pi2 with master-1 FPP code.  

Thanks for your work on this library.

dmjpixel, 2016-02-24:
A few others in the below forum were talking about the same experience - known pixel order being different on the Pi - thought this might help with background info.

http://falconchristmas.com/forum/index.php/topic,2100.120.html

jgarff, 2016-02-24:
The library already supports all strip types without code modification.  Instead of changing the ordering of the output buffer in the code directly, you can use the strip_type field in the channel structure prior to initialization.  Please see ws2811.h for details.  Any of the following can be used to comply with the ordering for your specific hardware.

WS2811_STRIP_RGB
WS2811_STRIP_RBG
WS2811_STRIP_GRB
WS2811_STRIP_GBR
WS2811_STRIP_BRG
WS2811_STRIP_BGR

dmjpixel, 2016-02-24:
Hi Jeremy - thanks for your review.  I reviewed the info in ws2811.h and I agree it supports the various strip types.  The issue is in the code area I suggested a change for; the order of the code is swapping back the R & G channel into a G & R order (or A-B to B-A to try and take the RGB out of the my explanation). 

For example and as designed, if the order selected by a user is GRB the routine will "force" the order into R (G) - G (R) - B (B).  The issue arises when the code writes the buffer back in G (R) - R (G) - B (B) order, which was most likely a remnant of previous hardcoding for a old WS2811 GRB order.  With the new method to accept any order of string the output to build the array needs to be switched.

This is a little hard to describe so I apologize for any confusion.  Thanks for your review.

jgarff, 2016-02-24:
Oh okay.  Thanks for the additional explanation.  I'll review the code more closely a bit later to make sure I'm understanding the issue properly.  Thanks for the changeset.  Much appreciated.

dmjpixel, 2016-02-24:
Absolutely - as you can see I am new to Github so I apologize in advance if I did not do something correctly in this process.  To say it a slightly different way - my theory is the output is built incorrectly no matter which string option is selected; the 1st two positions will always be swapped incorrectly when written.  Thanks again for your time to review.

dmjpixel, 2016-02-25:
While you are reviewing I noticed these discrepancies in the ws2811.h file.  This is not part of the current issue or my proposed fix but thought I would bring it up to you.  I would be happy to prepare a fix if you agree it needs adjusted.

The value for the GBR option below should be 0x080010 and the value for the BRG option below should be 0x001008 (I think...)

#define WS2811_STRIP_RGB                         0x100800
#define WS2811_STRIP_RBG                         0x100008
#define WS2811_STRIP_GRB                         0x081000
#define WS2811_STRIP_GBR                         0x001008
#define WS2811_STRIP_BRG                         0x080010
#define WS2811_STRIP_BGR                         0x000810

dmjpixel, 2016-02-28:
I created a detailed explanation (see attached) which hopefully helps describe the issue as well as how I came across it along with the proposed fix.  My apologies in advance for breaking this down into pretty simple details (and lengthy)...I needed that for my own sanity.  The coding itself is fantastic.  Let me know if I am missing something or if I should take this to the original contributor of this code section.  Thanks!

[RPI_WS281X_Detailed_Explanation.pdf](https://github.com/jgarff/rpi_ws281x/files/150011/RPI_WS281X_Detailed_Explanation.pdf)

jgarff, 2016-02-28:
Great, thanks.  I was just starting to look at the request.  Sorry for the
delay.  It's been a fairly busy week.

dmjpixel, 2016-02-28:
No worries at all...I was not able to describe the issue very well in my original comment so it has been bugging me and I wanted to provide something better...found the time this morning.  I appreciate your time and comments.

jgarff, 2016-02-28:
Looks good, thanks.  Would you mind adding your fixes to the string defines as well and send another pull request?  I'd like to make sure you get credit for those changes as well.  Additionally, I think the commit comment should warn users that this change will impact application code where they had to work around this issue previously by choosing a incorrect string type.  Users should be aware that their code will require a type fix upon update.  Thanks again!

dmjpixel, 2016-02-28:
Thank you.  I just added the 2nd commit and updated comments in pull request per your suggestion.  Assuming I did it correctly it should be in the original pull request.  Please let me know if I need to retract and create another.  Thanks again for your review.

dmjpixel, 2016-03-01:
Issue was resolved with merge of fix on 2/28.  Thank you.

closed 2016-03-01 by dmjpixel