jgarff/rpi_ws281x #64: Main.c hooks the worng signal

rurwin, 2016-02-28:
Sorry, not set up to do pull requests.

main.c:
-     sigaction(SIGKILL, &sa, NULL);
+    sigaction(SIGINT, &sa, NULL);

There's no difference in behaviour just doing the fini, but I added code to turn all LEDs off, and that only works with this change. SIGINT is " Interrupt from keyboard", ie ctrl-c. SIGKILL is "Kill signal" and is the backstop when all else fails; " The signals SIGKILL and SIGSTOP cannot be caught, blocked, or ignored." See man 7 signal.

jgarff, 2016-02-28:
Good catch.  Thanks.

jgarff, 2016-02-28:
Fixed in commit 30a63e7b7028ecc17578bef927c6406606660f0e.

closed 2016-02-28 by jgarff

jgarff, 2016-02-28:
Please also see commit 8d63a6c2fb870f11d20143a60bf7cff605cb0299.  I added the SIGTERM signal hook in case the user runs this program in the background and terminates with the kill command.

rurwin, 2016-02-28:
Nice quick fix, thanks.