jgarff/rpi_ws281x #68: Not able to light any LEDs on Pi3

Tetrikus, 2016-03-12:
Hello. First of all, thanks for your work to provide this! I tried to test this out on a Rapberry Pi 3. This is the configuration I use: I connected the strips directly to 5V and to the GPIO pin 18.

```
# LED strip configuration:
LED_COUNT      = 150     # Number of LED pixels.
LED_PIN        = 18      # GPIO pin connected to the pixels (must support PWM!).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 5       # DMA channel to use for generating signal (try 5)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
```

When I run ```strandtest.py```, it is running, but nothing is happening to the LEDs. Can anyone confirm, that this should work on the Pi3 as well?

Thank you so much for your help in advance!

jgarff, 2016-03-12:
I've confirmed that it runs fine on my Pi3.  If the program is executing without error, it's likely that there is a problem in your hardware configuration (level shifter, missing ground, etc.).  Another common problem is due to audio playback.  This library uses the same PWM that is used for audio.  If you want to support both audio and LEDs, you'll need to add a USB audio device instead.

closed 2016-03-12 by jgarff

marbaque, 2017-11-07:
Are you running the program from termina using "sudo". I happened to me that I was trying to open the  file directly with python3, and it doesn't work that way in this case.