jgarff/rpi_ws281x #72: How to determine free DMA channels ?

penfold42, 2016-03-20:
It seems that differnt kernels have different ways to report in use DMA channels.

The kernel is also told in its boot parameters which DMA channel the GPU wants to use.

Does anyone know a complete reliable way to determine this ? Googling around doesn't seem to provide a nice concise approach here.

Thanks !

myoung008, 2016-04-26:
DMA channel 5 is the only one that can be used by the hardware to run the PWM generator.  That's the only one that's interesting for rpi_ws2811, and it's required.

I'm not familiar enough with DMA to know if the kernel has a reporting interface, but you could mmap the hardware as done in this library and watch to see which channels are actually set up and in use.

jgarff, 2016-04-29:
I looked into this some time ago.  Unfortunately there wasn't a good userspace interface to determine DMA usage.  I think the proper way to solve this would be to implement this library as a kernel module.  In that case the DMAs are managed by a kernelspace driver.  I've wanted to do something like this for a while, but the downside is that it makes it more cumbersome to build and deploy and requires having kernel headers installed.

closed 2016-04-29 by jgarff