jgarff/rpi_ws281x #75: Permissions issue with library

miragevideo, 2016-04-01:
Hi All,

I am trying to run the python script from a PHP shell_exec command. I get errors that I think are related to the permissions from the loaded files in the main script.  I tried changing permissions to 777 on all the script files but I am still seeing the issue.  My guess is that I missed some files.  Can you let me know which files need execute permission or if there is something else I missed?

thanks

jeff

penfold42, 2016-04-01:
Is selinux enable ?

If so, try disabling it with "setenforce 0"

miragevideo, 2016-04-01:
Where do I see that property?  thanks

penfold42, 2016-04-01:
getenforce

usefulthink, 2016-04-02:
As for SELinux, this might help: http://selinuxforum.com/how-to-enable-or-disable-selinux-t15.html

Another question would be: are you running the php-script as root? At least the python script should iirc always be run as root (because it needs to access raw memory).

jgarff, 2016-04-29:
I haven't seen any updates on this issue in a long time.  Generally permission issue are distro/install specific.  This library doesn't do anything specific with permissions other than needing to be run as root.  I'm going to close the issue.  Please re-open if you need further assistance.

closed 2016-04-29 by jgarff

miragevideo, 2016-07-18:
Hello Again,

I am finally able to spend some time on this issue.  What I am seeing is that the code starts to run from my PHP->Python call.  It fails along the way, I am assuming, due to permission issues with some of the modules.
 If I run the php script from the command line I see a message like this:

failed to create mailbox device
: Operation not permitted
Traceback
from strip.begin()
file"/usr/local/lob/python2.7/dist-package/rpi_ws281x-1.0.0-py2.7-linux-armv61.egg/neopixel.py" line 102 in begin
runtimeerror: ws2811_init failed with code 1
segmentation fault

Do you think the selinux change will resolve this?  
My Dist does not have getenfore/setenforce commands so I will need to edit the /etc/selinux/config and reboot.  I have read about relabelling the file system.  Is this all part of the process?

Thanks for any help.

Jeff