jgarff/rpi_ws281x #76: RPI3 Error - *** Error in `python': free(): invalid pointer: 0x01cc4010 ***

dharvey24, 2016-04-03:
I've got a RPI3 and my code executes fine and runs the knight rider green and blue leds across a neopixel 8 pixel stick.  But at random times it generates this error message- 
*** Error in `python': free(): invalid pointer: 0x01cc4010 ***

any ideas?

dharvey24, 2016-04-03:
okay just tested the examples and they work great so i've managed to amend them.  So confirmed works on the RPi3.

I was using this code from here http://blog.whatgeek.com.pt/2015/04/raspberry-pi-and-adafruits-neopixel-stick/ - this still produces the error above.

SlySven, 2016-04-03:
I just read that article and I am concerned about something written there:
> The wiring of the Neopixel stick is straight forward. Connect GND (any GND of the stick – it as two)to Raspberry PI GND, DIN to any PIN, in my case, PIN 18, and 5VDC to 3.3V (click image for a bigger version)

:warning: :skull: **The person there has connected the *5V* supply on the stick to the *3.3V RPi rail* - that cannot be right IMHO.  It seems the device will drive all 8 of the WS8212 LEDs (and any more chained ones) from that supply and there is a *very real danger* that this will overload the RPis internal 3.3V supply!** :skull: :warning:

dharvey24, 2016-04-04:
Yep I saw that too and didn't make that mistake. I've got it connected to the 5v supply.

jgarff, 2016-04-30:
There seem to be a few problem with the setup in this writeup.  The one previously mentioned, regarding the current draw on the Pi's 3v3 rail, and another regarding the required voltage for both signalling and power to the LEDs.  Assuming you don't smoke the 3v3 regulator/inductor on the Pi, the LEDs will not illuminate as brightly (especially the blue).  Worse though, it may work only intermittently as the PWM signal voltage may not reach the high threshold reliably enough.  I suspect this would result in work fine for some, and maybe not at all for others, depending on the part.  Users should include a level shifter (3v3 -> 5v0) in the signal path.

closed 2016-04-30 by jgarff