jgarff/rpi_ws281x #77: ODROID support

peel, 2016-04-03:
Although I am no C dev, I was looking into running the library on ODROID C2. 
The pins are compatible, however there are some key differences. Starting with AArch64 architecture (coming to RPi3 sometime soon), HW naming scheme
```
Processor	: AArch64 Processor rev 4 (aarch64)
processor	: 0
processor	: 1
processor	: 2
processor	: 3
Features	: fp asimd crc32
CPU implementer	: 0x41
CPU architecture: AArch64
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: ODROID-C2
Revision	: 020b
```
And kernel version
```
[root@hell4 python]# uname -r
3.14.29-16-ARCH
```
I quickly tried to fall back to running as RPi2, yet faced the issue of lacking /dev/vcio and the /dev/mailbox  failed to be read (whereas creation worked fine). Any ideas how can I follow with that?

jgarff, 2016-04-04:
This library doesn't support the ODROID.  It only works on specific Broadcom processors present in the Raspberry Pi as it requires special hardware logic provided by the chip.  Unless there is a ODROID based around the exact same Broadcom parts used in the RPi, it won't work.

closed 2016-04-04 by jgarff

guilhermegm, 2016-05-16:
@jgarff 

I'm using Ubuntu Mate 16 on rpi 3 and I got the same error on the lastest commit.

Thanks.

puccaso, 2018-11-26:
im actually getting the same error on rpi3b with 64bit kernel/32bit userland

```
merateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[4219:4237:1126/212758.099440:ERROR:sandbox_ipc_linux.cc(438)] /dev/vcio: ioctl: Inappropriate ioctl for device (25)
usrvcsm: [vcsm_malloc_cache]: [1] [mmal_vc_port buffer]: NULL size or invalid device!
usrvcsm: [vcsm_vc_hdl_from_hdl]: [1]: invalid device or handle!
usrvcsm: [vcsm_lock]: [1]: invalid device or invalid handle!
mmal: mmal_vc_shm_alloc: could not allocate 122880 bytes of shared memory (handle 0)
mmal: mmal_vc_port_payload_alloc: vc.ril.resize:out:0(bgra): failed to allocate 122880 bytes of shared memory
mmal: mmal_pool_initialise_buffer_headers: failed to allocate payload 0/16
[1:19:1126/212758.163385:ERROR:mmal_video_decoder.cc(1340)] ### mmal_port_pool_create failed: 16*122880
```