jgarff/rpi_ws281x #81: Some images (osmc & retropie) have something that conflicts

markwal, 2016-04-10:
Sorry, this is probably not an issue with rpi_ws281x, but I'm hoping you can help me track down the problem so I can do a a PR to the right place.

Using a logic analyzer, it looks like something in those images (osmc and retropie) have something that causes the timing to be off. 

On a regular jessie-lite image, I get an 800 kHz wave form with a zero looking like 420ns on and 830ns off (+/- the resolution of my LA).

On the broken images, I get a ~75KHz wave form with a zero looking like 400ns on and 12900ns off and a one looking like 4200ns on and 9200ns off.

If I crank up the target frequency in strandtest from 800000 to 1600000, sure enough, it cuts those times in half.  And again, 1600000 to 3200000.  However, going to 6400000, strandtest gets very unresponsive (hard to cancel) and it looks like I don't get a good signal, so I'm exceeding some workable threshold.

Any idea what the guys building the images can be doing to make the timing ~10 slower?  It doesn't look like it is the audio device conflict (I took care to disable and when strandtest isn't running, there's no signal at all).  I've also tried different DMA channels and the result looks the same.  Perhaps it is something to do with interrupts and/or scheduling?

penfold42, 2016-04-10:
I have some notes from the weekend that I will post here.

The conclusion so far seems to be a that :
 if you boot OSMC with an HDMI monitor it works.
If you boot with no HDMI it doesn't.

The kernal versions are very different - from memory it's 4.1.3 on Jessie and 4.6.6 on the latest OSMC

markwal, 2016-04-11:
Huh.  I wonder how I haven't tried that yet.  Sure enough, that works also with the RetroPie image.

It would suggest a conflict with audio PWM since with an HDMI connected it will send the audio there instead of the analog port, but... something still doesn't make sense.  Manually forcing the audio to HDMI doesn't help.

penfold42, 2016-04-11:
My guess is the timers are setup differently by the kernel, GPU or boot loader and this code works fine with this assumed state.

jgarff, 2016-04-29:
You might also try using a different DMA if you haven't already.  I haven't seen any updates on this issue in a few days, so I'm going to close it.  Please re-open if this continues to be a problem.  Thanks.

closed 2016-04-29 by jgarff

penfold42, 2016-04-30:
It is still an issue with OSMC.

If you boot won HDMI it works fine
If you boot with no HDMI the timing is all wrong.

After booting, if you change the video output  with tvservice it makes no difference - the LEDs either work fine or or don't based on how you booted

markwal, 2016-04-30:
Different DMA doesn't appear to change anything. When booting without HDMI plugged in, the timing is approx 10x too slow (the LOW time is 10x). And the same result with both PWM channels and all of the four pins.

Maybe the analog sounds driver has a bug that even when it is unloaded it leaves behind some detritous? Or maybe it sets some register that rpi_ws2812 doesn't reset.

jgarff, 2016-04-30:
You could try blacklisting the audio driver kernel module.  That would
ensure that it can never be loaded and conflict.  I've never used OSMC, but
typically you would do that in /etc/modprobe.d/blacklist.conf.