jgarff/rpi_ws281x #83: Inversion color problem python

maxouille, 2016-04-11:
Hi,

I encouter a problem with the led colors in python. It should be RGB order but when I try to color leds here is the colors I need to use : 
red = Color(0, 255, 0)
green = Color(255, 0, 0)
blue = Color(0, 0, 255)

However, in neopixel.py the shift in the first function is good in my opinion. 

Do you have any idea about this problem ? 

Best regards

junes, 2016-04-15:
First of all I want to thank you for the great work, which makes it so easy to play around with great stuff! 
I just ran into the same problem with the latest version. Everything worked as it should for the last  months until I installed it from scratch on a new raspi two days ago. Green and Red are swapped now.

penfold42, 2016-04-15:
Did you explicitly specify the strip type ?

There were some changes to this over the last few months.

junes, 2016-04-16:
OK, thanks. After some search I finally found it: pydoc _rpi_ws281x did the job, as I did not know where to find the valid values for the different types.

    WS2811_STRIP_BGR = 2064
    WS2811_STRIP_BRG = 4104
    WS2811_STRIP_GBR = 524304
    WS2811_STRIP_GRB = 528384
    WS2811_STRIP_RBG = 1048584
    WS2811_STRIP_RGB = 1050624
    WS2811_TARGET_FREQ = 800000

Maybe you could add those to your strandtest.py example or somewhere else in the documentation.

penfold42, 2016-04-16:
Not my code ! Great idea tho - pull requests are alway appreciated

msurguy, 2016-06-06:
I'm having the same problem. Can anybody tell me  how to use these values in Strandtest.py?

hug33k, 2016-07-09:
Hi,

I got the same issue with my ring.

You can fix it by adding values found by @junes like this:
````python
from neopixel import *
import _rpi_ws281x as ws
ring = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, strip_type=ws.WS2811_STRIP_GRB)
````

closed 2016-07-10 by jgarff