jgarff/rpi_ws281x #84: ws2811_init failed with code -1

ppfenninger, 2016-04-13:
I followed the installation, but whenever I try to run strandtest.py example I get this error. 

File "strandtest.py", line 83, in <module>

strip.begin()

File "build/bdist.linux-armv7l/egg/neopixel.py", line 100, in begin

RuntimeError: ws2811_init failed with code -1

penfold42, 2016-04-14:
What pi do yoi have ?

Can you please post /proc/cpuinfo ?

BogdanBerg, 2016-04-14:
I'm hitting exactly same issue:
  File "strandtest.py", line 83, in <module>
    strip.begin()
  File "build/bdist.linux-armv6l/egg/neopixel.py", line 102, in begin
RuntimeError: ws2811_init failed with code -1

This is on Pi Model A+

And /proc/cpuinfo:

processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 2.00
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 0015
Serial          : 0000000022058e83

BogdanBerg, 2016-04-14:
This is interesting, I just pulled and looks like it's working now! I mean, I can run strandtest.py without hitting the errors, I have yet to connect the LEDs.

Looks like this is the fix:

commit b3cfb853fc4ffdb15c4899c8d5d853b662d45616
Author: Mackenzie Boyd <macboyd+github@gmail.com>
Date:   Sat Apr 9 19:43:42 2016 -0700
    Additional Model A+ Revision (hwver) 0x15

Thank you Mackenzie! :) :+1:

ppfenninger, 2016-04-14:
I am using a raspberry pi 2. 

processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2709
Revision        : a01041
Serial          : 0000000031a6ca8e

jgarff, 2016-04-14:
From your proc information, it looks like your device should be supported.
Here's a couple things to check:

- Make sure you're running the program as root.
- Make sure you have the Broadcom videocore driver is in your kernel, and
that the /dev/vcio device file is present.  This should already be setup on
most distros such as raspbian, but some distros are known to be
problematiic.

Jeremy

closed 2016-04-30 by jgarff