jgarff/rpi_ws281x #89: Can you use RGBW?

HUGEwindow, 2016-05-04:
Is there a way to use RGBw neopixels?

penfold42, 2016-05-05:
Not yet #89

Do you own some already ? If so, what type and are you able to help test ?

HUGEwindow, 2016-05-05:
yes, I have a neopixel 144 strip. I am cut them up into segments of three. I am using a raspberry pi.

penfold42, 2016-05-05:
Do you know the specific LEDs on the strip ?

penfold42, 2016-05-05:
Looks like my branch might work...

https://github.com/penfold42/rpi_ws281x/tree/sk6812?files=1

Can you try this ?

It compiles for me, but hasn't been tested at all

HUGEwindow, 2016-05-05:
Sure let me give it a shot.  What do you mean by 'Know'?

HUGEwindow, 2016-05-05:
What url should I clone?

HUGEwindow, 2016-05-05:
That branch compiles, but do we need test that specifies rgbw?  Hard to know for sure whats happening with test and strand test

penfold42, 2016-05-05:
It should work for RGB strips as it stands.
Apps using it need to set the strip type to SK6812_STRIP_RGBW and then it should use the RGBW protocol.

I'll have a closer look tonight

HUGEwindow, 2016-05-05:
ya rgb is good, ok I will set strip type and test.

HUGEwindow, 2016-05-05:
Any luck with this? Do you know any code that can drive RGBw?

penfold42, 2016-05-06:
I think so !

I've just pushed a change to main.c to specify the strip type.

Can you run it and see if my works ?

I've orders a strip myself but it will take a couple of weeks to get here so I can only guess test !

Thanks

HUGEwindow, 2016-05-06:
Ok, I will test. Not sure I have those leds in toe today. If not first thing this evening. I will let you know and might need your guidance.

penfold42, 2016-05-06:
I've also added SK versions of strandtest.py and lowlevel.py.

again, not tested on real LEDs

HUGEwindow, 2016-05-06:
Works! Thanks this great. Be cool to commingle them. Thank you. I really appreciate it. I will let know how it performs. Let me know if you have questions.

HUGEwindow, 2016-05-06:
Is it possible to set the brightness to brightness to individual pixels?

penfold42, 2016-05-06:
Ezxellent !

Yes - yoi can set each led to any desired colour or brightness 

The c and Python sample are pretty basic - you can see the colour tables they use and modify them 

I need to try actually using the white LEDs. At the moment the examples still just use RGB ignoring W

b3joel, 2016-05-08:
Got it working here too without any hickups. Can't identify the exact RGBW leds i have though.
But, THANK YOU Penfold42!

penfold42, 2016-05-08:
Cool !

So now we have basic support working, we need to try writing and effect that uses the W led.

From memory the highest 8 bits of the uint32 that represents LED colour is sent to the white channel.

So 0x00FFFFFF is white using the RGB LEDs and
0xFF000000 is white using just the W LED.

penfold42, 2016-05-09:
I committed some updates to my branch.

There a replacement for test called..... newtest!

This takes cmd line arguments for the strip type, height, width, gpio, dma channel

It also has a different colour table if you're using rgbw led strips that (hopefully!) will exercise the W channel

b3joel, 2016-05-10:
Will test this evening and report back

b3joel, 2016-05-10:
Just wanted to add. It doesn't work with a RPI 2. I have both the RPI 2 and 3. Cloned the SD-cards, but on the RPI2 it just doesn't work.

b3joel, 2016-05-10:
Everything still works, but it's above my level script something to get the white to work.
Was trying something with: strip.setPixelColor(0, Color(  0,  0,  0,255))
But the fourth entry is not accepted.

penfold42, 2016-05-10:
I need to add new methods to the neopixel library to support 4 colours.

Does your pi2 have an HDMI display attached ?
I find with OSMC, that this only works with an hdminattached display.

With the pi1 and pi3, does newtest -s rgbw work ?

b3joel, 2016-05-14:
I get a 'command not found' when using newtest. Even when using sudo.

tonydew, 2016-05-18:
Sorry to hijack this issue, but I am unable to post one over at @penfold42 fork: https://github.com/penfold42/rpi_ws281x.git 

I followed the instructions given here (pull, scons, python setup.py etc) and I get the following error when I try to run `examples/SK6812_strandtest.py`

```
Traceback (most recent call last):
  File "examples/SK6812_strandtest.py", line 20, in <module>
    LED_STRIP      = ws.SK6812_STRIP_RGBW
AttributeError: 'module' object has no attribute 'SK6812_STRIP_RGBW'
```

Running `sudo ./newtest -s rgbw` runs *something* but I'm not sure what is supposed to show. Its just a bunch of cycling color LEDs, but it's hard to tell if any of them are the pure white LED or not.

Any idea what might be up? This is on a Raspberry Pi 3, with up to date Raspian and such.

BogdanBerg, 2016-05-18:
Thought I'd share, as a data point: I was able to use code from the SK6812 branch just fine (in C though, haven't tried in Python).

Everything works like a charm, with the only weirdness being: R and G values are swapped. I'm using "Mokungit SK6812" LED strip from Amazon with RPi 3.

Thank you penfold42 for your great work! :+1: :)

b3joel, 2016-05-18:
Having no problems with the strandtest. R and G are swapped but this is
also specified in the adafruit neopixel guide.
Also on RPI 3.
@Tony Dew, make a new folder and git the correct branch. Then navigate to
the examples folder and sudo python SK6812_strandtest.py

penfold42, 2016-05-18:
@tonydew did you have a previous build installed previously?

I'm travelling so dont have time to troubleshoot this properly right now but some tthoughts that might help:
1) I remember when I was getting this working that I seemed to have to fight the python package installer to overwrite the existing package.
2) I also recall having to find out how to use swig to rebuild the some of the python stuff. 
It is controlled by rpi_ws281x.i

tonydew, 2016-05-18:
@penfold42 thanks for you help on this. Yes, I did have the `jgarff/rpi_ws281x` build installed and running previously in `~/rpi_ws281x`

Yesterday, I tried installing your *sk6812* branch in `~/test` by following your exact instuctions from earlier in this issue:

```
git clone https://github.com/penfold42/rpi_ws281x.git --branch sk6812
cd rpi_ws281x
scons
cd python/
python setup.py install
python examples/SK6812_strandtest.py
```

That's where I ended up yesterday.

**HOWEVER...**

After the hint about removing `/usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg` I just went through the following process:

```
sudo rm /usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg
cd ~/test/rpi_ws281x/
scons
cd python/
sudo python setup.py install
sudo python examples/SK6812_strandtest.py
```
...and BOOM! It seems to work now.

I had to modify neopixel.py and SK6812_strandtest.py, and also run `sudo python setup.py install` again, but it works and uses the white LEDs.

I'll fork your SK6812 branch in a bit and do a PR with my changes. Also, I'll upload a video of it somewhere to show the white LEDs in use.

Thanks!

tonydew, 2016-05-19:
Works like a hot damn! Thanks. Pull request for your SK6812 branch @penfold42 - https://github.com/penfold42/rpi_ws281x/pull/4

penfold42, 2016-05-19:
I really hope  my fork will be merged into this repo - I don't want to maintain something that's 99.99% the same as something else !

Thanks for your testing guys - I still have not received the strip I ordered so have never seen the joy that is white light !

tonydew, 2016-05-19:
Potato-quality video of it running through full brightness Green, Red, Blue, **WHITE!!!**, Composite white (R+G+B) and finally all channels (R+G+B+W). My version has warm white to match the rest of the lighting in this project, and it's quite yellow compared to the very blue composite white:

https://www.youtube.com/watch?v=zfTEBqSP6dk

Was trying to get a rough estimate of current draw. Details are in the description. I'll have a better quality video tomorrow once I've gotten some sleep. :)

penfold42, 2016-05-19:
Thanks for sharing - it's great to see some of my code actually works with real hardware rather than looking at a berserk 2812 strip and hoping !

If you're looking for something else to to drive it check out Hyperion - it uses this library as one of the options

tonydew, 2016-05-19:
Thanks for your work and the merges. I've just sent another PR that has LED color ordering fixed and working for both RGB and RGBW strips, even in a multi strip setup.

This code was absolute gibberish to me at first, but after a few hours in it, it's all starting to make sense. Or at least much of it is. This is the first time I've ever worked in C, and almost the first time I've wroked in Python. So many little land mines tripping me up. :)

penfold42, 2016-05-22:
I've just pushed some minor changes to my branch for the predefined strip types.

@jgarff - are you happy with our approach ?
Will you merge a PR when I create it ?

jgarff, 2016-05-30:
Absolutely!  I've just taken your code into the "sk6812" branch on the mainline repo.  Let's use this branch for a couple weeks for regression testing.  As I also mentioned in your pull request, I'd like to replace test with newtest as well.  The command line options are great.  Thanks to @penfold42 and @tonydew for these new features.  Great work!

Sorry for my slowness in getting this in.  Life has been a bit crazy lately, and I had to order a RGBW array to test with.  At this point, I'm going to keep this issue open so we can discuss any ongoing changes and/or issues.  I appreciate everyones help in testing this before we merge this into the master branch.

penfold42, 2016-05-30:
Excellent - thanks

jgarff, 2016-06-18:
I've done some updates on the sk6812 branch, which include changing newtest to test.  If you all would re-test to make sure nothing is broken I would greatly appreciate it.  Thanks.

closed 2016-06-18 by jgarff