jgarff/rpi_ws281x #93: ws2811_init failed with code -1  Again

thelasthope4U, 2016-05-12:
First I apologize for bringing this issue up on the board again, this is my first post.   I surveyed the other posts but do have a different view on this and wanted to get some feedback.  First I need to disclose I  am new to raspberry PI so I may have some problems out of the gate.

I have a model b+, below is my CPUINFO

processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 2.00
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 0013
Serial		: 00000000b475e151

I read this as Hardware revision 13 since iti s 0013 not 0x13.  According to the table in rpihw.c this would match the entry for a model B
   //
    // Model B
    //
    {
        .hwver  = 0x0d,
        .type = RPI_HWVER_TYPE_PI1,
        .periph_base = PERIPH_BASE_RPI,
        .videocore_base = VIDEOCORE_BASE_RPI,
        .desc = "Model B",
    },

but model B+ has .hwver = 0x10 ?

In either case, when executing rpi_hw_detect in rpihw.c, there is no match detected at all.   This is where I am getting lost.  if the following line 
            rev = strtoul(&substr[1], NULL, 16);  // Base 16
were changed to 
            rev = strtoul(&substr[1], NULL, 10);  // Base 10
then I can get a match in the loop for Model B as shown above, but I don't know if that is desired.

So if someone can point me in the right direction I would appreciate it.  I installed the O/S using the Noobs download package from the raspberrypi.org website.

Hopefully this is enough for those experienced here to see the error in my ways.

Thanks

penfold42, 2016-05-12:
its hex, not decimal - theres a list here:
http://elinux.org/RPi_HardwareHistory

maybe we need to add another test case

penfold42, 2016-05-12:
try my branch - i've just pushed the change to it.
it compiles but i cant test it
https://github.com/penfold42/rpi_ws281x

thelasthope4U, 2016-05-12:
Thanks for such a quick response.  I was able to build and execute without the error.  I did not see the LEDs light up yet but this is the first time the code executed.  I will have to review my hardware setup.  To be honest I did not have a 74HCT125 available but found a 74125 (yes I have been around that long) in the parts bin.  I will have to break a scope out tonight and see what is going on.  I have to break away for now and go earn some money.  The hobby has to wait till tonight.

Thanks again, I will probably be back again with more questions later.

Gadgetoid, 2016-05-16:
This will fail again for the Pi Zero v1.3, which has a revision number of 0x900093.

penfold42, 2016-05-16:
pull request created for .hwver 0x13 and 0x900093. 
#94

jgarff, 2016-05-21:
Pull request committed, closing issue.

closed 2016-05-21 by jgarff

thelasthope4U, 2016-05-22:
I was able to complete my tests today, this fix worked great, thanks.