jgarff/rpi_ws281x #96: neopixel.py Color function has red and green flipped

pacmandu, 2016-05-30:
Hey, I noticed you have have some logic flipped on "https://github.com/jgarff/rpi_ws281x/blob/master/python/neopixel.py#L11"

The red and green values are flipped.

The line should read "return (red << 8) | (green << 16) | blue" and not "return (red << 16) | (green << 8) | blue"

I'm on a Raspberry Pi 3 with the 74AHCT125 IC chip and followed the guide on Adafruit here: "https://learn.adafruit.com/neopixels-on-raspberry-pi?view=all#software"

Thanks for all your work on this library.

pacmandu, 2016-05-30:
Didn't see the previous issue on red and green being swapped. If you are already working on a fix then you can ignore this issue. Thanks!

penfold42, 2016-05-30:
Not a bug - you need to set the strip type appropriately.

The default is strip_type=ws.WS2811_STRIP_RGB):

Yours need GRB

pacmandu, 2016-05-30:
Oh gotcha. Thanks for the quick reply!

closed 2016-05-30 by jgarff

charliesneath, 2018-06-05:
Having a similar problem, but not sure why my settings to change to `GRB` are not taking.

I'm updating the setting in main.c to be:

`#define STRIP_TYPE WS2811_STRIP_GRB`

@pacmandu Where did you make your change?