jgarff/rpi_ws281x #97: RPi Detection fails when warranty bit is set

andbey0nd, 2016-05-30:
At line 260 of rpi_ws281x/rpihw.c a change is needed to detect properly "Revision" string for Raspberries with warranty bit set (i.e. after use of force_turbo or over_voltage settings).

`rev = strtoul(&substr[1], NULL, 16); // Base 16`

must be changed to:

`rev = (strtoul(&substr[1], NULL, 16) & 0xffffff; // Base 16`

For example my RPi2's original Revision was "0xa01041", but after using force_turbo (voiding the warranty) it became "0x**1**a01041". The same thing happened on my RPi3, where the original revision was "0xa22082" and it became "0x**2**a22082".

I compiled hyperion with this change and now everything works as it should.

jgarff, 2016-05-30:
Good to know, thanks.

jgarff, 2016-06-18:
Fixed in aadcebabd5c122190d429b52e58e048850a7fb12 on the sk6812 branch.  Please re-test and let me know if you run into problems.  Thanks.

closed 2016-06-18 by jgarff