jgarff/rpi_ws281x #98: Any attempts to build a version for Node?

che-effe, 2016-06-01:
I'd like to take a stab at building a node version. Do you currently have / are planning to have some work done to tackle this?

usefulthink, 2016-06-02:
Already did that: https://github.com/beyondscreen/node-rpi-ws281x-native
Would love to see people using it or maybe helping out with development :)

che-effe, 2016-06-02:
Excellent! I might play around with this today.

usefulthink, 2016-06-02:
Great! I didn't have a lot of time for this lately, so it's now well behind the recent changes in this library, but if you're using standard ws2812 LEDs and raspberry-pi 1 or 2 everything should be fine.

jgarff, 2016-06-02:
If you're changes are compatible, and you're willing to update to the
latest, I'd be happy to review and incorporate them into the mainline.  If
you do, please use the sk6812 branch.  That's where the latest updates are
being staged.  Thanks.

usefulthink, 2016-06-02:
@jgarff I wasn't sure you would like it, so I kept it in a separate repository :) (using your reporitory via git submodules).

che-effe, 2016-06-02:
@usefulthink, I'm working with Edison with mini breakout, may try to build it to run on that device.

usefulthink, 2016-06-02:
Oh, that will be a problem as this library is very specific for the Broadcom SoCs built into the Raspberry Pis and will not work with other platforms.

che-effe, 2016-06-02:
Gotcha, I'll try a separate board for LED control.

jgarff, 2016-06-07:
Closing this issue as resolved.

closed 2016-06-07 by jgarff