jgarff/rpi_ws281x #100: rPi can't keep up?

DriesOeyen, 2016-06-14:
Hi all,

This might be a stupid question (rPi newbie here), but I need some help here. I'm trying to control a 60-LED Neopixel strip from a Raspberry Pi B+, but there seem to be some timing issues out of the box. It seems like the rPi can't keep up, resulting in random light patterns. [Video here.](https://goo.gl/photos/4wABaHdhoYpLd2cj7)

Here are a couple of things I noticed:

- Running any of the demos (C and Python) results in chaotic flashing. Tried switching GPIO pins (18 and 12) and DMA channels, same result. 
- A modified version of strandtest.py in which I attempt to control a single LED yields interesting results: the strip starts flashing as usual, then after a couple of seconds the first LED switches to displaying a smooth rainbow animation. I believe that this is the effect I should be seeing. However, when I up the LED count to 2 all it does is flash incessantly again.
- At this point, I started to think about what could be slowing down the output on the GPIO pin. I had been working on the rPi over SSH from my laptop and had also set up a VNC server, which was active but not connected. Upon disabling these,  a change happened that was in line with my previous observations: the flashing pattern I had been seeing before consisted mostly of R, G or B colored LEDs. The new pattern consisted mainly of white / differently colored LEDs. This appears to indicate that the output frequency on the GPIO pin had gone up.

I'm not sure how to proceed. The LEDs are still flashing, so clearly the rPi still can't keep up. I'm running this on a fresh Raspbian install. Any suggestions and/or blatantly obvious things I'm missing here?

penfold42, 2016-06-14:
Do you have an HDMI screen attached ?

If not, can you try rebooting with one attached ?

electrokean, 2016-06-15:
Are you using a level converter on the GPIO pin?
See the Wiring section of the Adafruit NeoPixel article linked in the Wiki page

Are you using the same power supply for the LEDs and the RPi?
Voltage fluctuations due to CPU processing or LED activity could affect data pin signal integrity, especially if you're not level shifting.  If possible, use two separate power supplies, but with a common GND connection.

DriesOeyen, 2016-06-15:
@penfold42: in subsequent tests, I moved from using SSH to plugging directly into a HDMI screen. Same result. I don't see why this could make a difference, though?

@electrokean: Sorry, should've given more details. Here's what my setup looks like:

- Separate power supplies for the rPi and the LED strip.
- Grounds of the rPi and LED strip are connected.
- I'm *not* using a level converter, but I am using the diode method explained in that same article (same diode and everything). I can try the level converter method tomorrow night though, I'll post here when I have some results.

penfold42, 2016-06-15:
Did you power cycle the pi with HDMI plugged in ?
Something on the firmware configures the hardware differently and I've seen you situation with OSMC (which uses this library)

DriesOeyen, 2016-06-15:
Oh, interesting! I'm not sure if I power cycled, so I'll try that too. Will report back tomorrow evening.

jgarff, 2016-06-15:
Performance is usually not a concern.  Once setup, the signal is sent
without the CPU being involved.  It is also best to use a level shifter to
ensure a clean signal.  You'll also want to make sure you're not using the
Raspberry Pi audio as it conflicts in the hardware.

DriesOeyen, 2016-06-17:
Using a level shifter made no discernible difference in behavior, but connecting a HDMI display _at boot time_ totally did the trick! Thanks, @penfold42, I wouldn't have figured that out on my own. I was actually going to order another rPi to see if this was a hardware issue.

Also, I have no way whatsoever to explain this, but everything is now working as expected... even when the Pi boots without a HDMI screen connected. Exact same setup as the one I started out with. Ah well, at least it's out there now in case anyone else runs into this issue.

Thanks for helping me troubleshoot, everyone!

closed 2016-06-17 by DriesOeyen

penfold42, 2016-06-17:
Did you update the firmware or use a different distribution ?

For me OSMC had this issue but raspbian didn't

DriesOeyen, 2016-06-18:
@penfold42 No change to the software whatsoever. No distribution or even package updates, no reinstall of Raspbian. Just the act of plugging in a HDMI cable at boot time, as far as I can tell.

penfold42, 2016-06-18:
I've noticed that as long as it's plugged in at boot, that's all that matters.

You can then unplug it, change video outputs from the cmd line and all is good.

An interesting experiment would be force_hdmi

DriesOeyen, 2016-06-18:
@penfold42 Actually, my HDMI screen does **not** have to be plugged in at boot anymore now. Not sure why that is, but I'm happy since it was kind of important for my use-case.