jgarff/rpi_ws281x #102: Display Connection Affecting GPIO

AmphibiousLollipop, 2016-06-28:
Hey jgraff,

Im enjoying using this library for my projects, but i have been experiencing one issue I was hoping you could shed some light on. I am running 120 LEDs on a WS2821b ledstrip with a RPi 1 (all pretty standard stuff). When I have my HDMI display connected the strips is functional, however when I disconnect the strip flickers randomly. 

I assume that when the Pi notices it no longer needs to drive the display it alters something in the PWM clock. I am no expert here, so I was hoping you might be able to point me in the right direction?

Thanks in advance,
AmphibiousLollipop

penfold42, 2016-06-29:
Yep - I've seen this too. It seems dependent on the distribution and version too. 
When I last looked at this a few months back, OSMC would do this but raspbian was ok.

I suspect the firmware is setting up the hardware differently and this library is missing some timer register.

Does force_hdmi help ?

AmphibiousLollipop, 2016-06-29:
Hey penford42

Thanks for the reply. My first instinct was also to 'fake' the display's presence so i forced HDMI and all the same settings of display's EDID. Unfortunately it didnt help.

penfold42, 2016-06-29:
Id love to get all CSI on this issue but am away from my Pi's for a while so it's a bit hard to unplug HDMI!.

I think we need a bit of code to dump all the clock related registers so we can compare with and without HDMI.

mellena1, 2016-07-06:
Having the same issue. I honestly have no clue how any of the low level code works, haven't really looked into it at all. I may take a look tomorrow if I have time and see if I can make heads or tails of it. I'd really like to get this fixed, as having to have HDMI plugged in kinda ruins my project.

jgarff, 2016-07-06:
Which distro are you running?  I've never seen this issue, but I always use raspbian.  I suspect, as does @penfold42 from what I gather, that other distros have different kernels and/or clocking configurations which may cause problems.

mellena1, 2016-07-06:
Latest version of Raspian Jessie when I installed around 2 weeks ago. Just to provide more info from my end, when I have HDMI plugged in before I power the Pi on, the test works as expected with the rainbow animation. Without HDMI plugged in, the LEDs pretty much just blink randomly. In both scenarios I'm executing "sudo python strandtest.py" over ssh.

jgarff, 2016-07-06:
Are you making sure to tie the Pi's and LEDs grounds together?  Please be careful to avoid ground loops if using multiple power supplies.  I'm also wondering if your power supply is properly sized for driving the necessary current.  Thanks.

mellena1, 2016-07-06:
Grounds are all together, only one power supply. I'm using a molex connector from a computer power supply, and if what I've read is correct, the power supply can supply up to 11A on that pin. And a wiring issue doesn't seem likely to me, as its wired up the same both when HDMI is plugged in and when its not. The only variable of whether it is working or not is the HDMI cable.

jgarff, 2016-07-06:
I only wonder because I've seen cases where a power circuit is being completed through a alternate route, potentially HDMI.  In those cases (mostly USB), weird things happen when devices are unplugged, including flickering.

I'm not sure if this helps or not, but one thing I always do, is put a good sized capacitor right on the PI at the 5v0 rail which is shared with the LEDs.  I've personally run into issues, including flickering, when the power supply, even a good high current one, can't keep up with the instantaneous power draw of the LEDs.  A good sized cap (like a 220 or 470uF@10v0) can make a huge difference.

If you have access to a scope, I'd definitely like to see what the signals coming from the PI look like going into the LEDs.  This could really tell us if the signal quality is bad, or if the clock is indeed different.

mellena1, 2016-07-06:
You maybe right about the circuit being completed through HDMI. I just figured out that it works with the mini-HDMI to HDMI adapter plugged into the mini HDMI port on the Pi Zero but no HDMI cable plugged into the other end. Unfortunately I don't have access to any capacitors or a scope, I could pick some caps up tomorrow though. (Still relatively new to electronics tinkering so I don't really have any spare stuff laying around) 

With this info, where should I go from here? It must be something with the HDMI completing the circuit, but why would that be? Is it likely a minor wiring mistake on my part or something with the Pi? I could upload a picture of my setup if that helps. And why would it happen to myself, @penfold42, and @AmphibiousLollipop? Could it just be that we all coincidentally made the same mistake? Thanks again for the help, I'm definitely learning a lot over the course of this project :).

mellena1, 2016-07-06:
Just want to add as an update, if I have the adapter in on boot, start the python script, have it working perfectly, and then unplug the HDMI adapter, the script continues to run fine. So it may not be a circuit problem after all.

jgarff, 2016-07-06:
That's interesting.  It's really hard to say without a more concrete way to diagnose the power or signalling.  My gut feel on this one is that there isn't enough capacitance to keep up with the startup draw, so the voltage dips.  I wouldn't think that adding a HDMI connector would add enough capacitance, but I suppose it's possible.  I've seen stranger things.

At this point, I strongly suggest adding a cap to the 5v0 rail.  That might not solve it, but I think that's the best place to start.  As I mentioned before, I've personally seen similar issues (though I never tried the HDMI) when I didn't have enough power buffer.  Hope that helps.  Good luck.

mellena1, 2016-07-06:
Thanks for the help. I don't know if I even need a cap at this point. I'll do more testing tomorrow just to make sure that it is completely reproducible, but after taking the adapter out while the script was running, it works flawlessly without the adapter in ever. Really odd but, hey if it works, I'll take it.

As a side note, I'm working on a GUI program to be able to control the LEDs over USB from a host computer to make the LEDs inside my rig look cool haha. If you (or anyone really) is interested in it, I'll make it public once I get more progress done on it. Its nothing incredible as I'm still learning to code more complex things and more efficiently, but its pretty cool so far. You did most of the heavy lifting already to make my project even possible @jgarff, so thanks for that :).

jgarff, 2016-07-06:
Sure thing.  Glad you're finding the library useful.  Once your project is done, you should put it up.  It's always cool to see new projects and ideas!  Plus, it's fun!  Good luck.

penfold42, 2016-07-06:
The pin detects an HDMI connection by looking at one of the HDMI pins.

I suspect the leads (and your adaptor) both set this pin to indicate presence so you adaptor makes the pi think there's an hdmi connection.

We need a debug program to dump all the timer related registers both with and without HDMI to see what we're missing

Gadgetoid, 2016-07-13:
Could this be related to recent changes in Raspbian that cause the audio hardware to be in use by X, even if you are not explicitly playing audio?

See: https://github.com/jgarff/rpi_ws281x/issues/103

And: https://github.com/pimoroni/unicorn-hat/issues/56

Or, TLDR, try adding these lines in `/boot/config.txt`

```
hdmi_force_hotplug=1
hdmi_force_edid_audio=1
```

AughtNaughtZero, 2016-07-18:
Hello Group ... I want to 1) confirm that I believe the root of the issue described in this thread is related to the most recent Raspbian release and 2) thank you for the config.txt 'fix' described above.

I experienced the random flickering of LEDs this weekend on a project that has been running uneventfully since April _after_ updating Raspbian to the latest 2016-05-27 release (the project Instructable is documented [here](http://www.instructables.com/id/LED-Weather-Forecast/)). Though I could not be sure of the exact Raspbian version that had run uneventfully, I swapped the problem Pi out with another running version 2016-03-18 and the project behaved as expected ... after updating this second Pi to the latest Raspbian version, the issue returned.

jgarff, 2016-07-18:
Have you made sure to disable the audio kernel module after the upgrade?

AughtNaughtZero, 2016-07-18:
Thank you for the suggest jgarff ... I undid the /boot/config.txt edits suggested above and then blacklisted the sound card inside of /etc/modprobe.d as described [here](http://www.instructables.com/id/Disable-the-Built-in-Sound-Card-of-Raspberry-Pi/) and that fixed the issue as well.

AmphibiousLollipop, 2016-07-21:
Dearest @jgarff @mellena1 @penfold42 @AughtNaughtZero @Gadgetoid ,

First of all a huge thank you for all your efforts to figure this one out. My project now works flawless and i have learned a lot from this process. Adjusting the config.txt file with the following:

hdmi_force_edid_audio=1

solved the issue for me on on my RPi 1 as described in the first post of this thread. For later RPi other configuration is required or even disabling the sound card completely to ensure no unwanted signal distortion.

And i cant close without thanking @jgarff again for an outstanding library and his continued efforts to help.

See you all next time!

UrielGuy, 2016-07-30:
Hey.  I'm seeing this problem as well - seems like this library and analog sound don't play nicely together, at least not on my setup (RPi3, latest Jessie release). happy to help diagnose, as I sort of need these two together, I have a scope in hand if that helps.

Gadgetoid, 2016-07-30:
@UrielGuy it impossible to use both analog audio and this library together, it would be akin to plugging two USB devices into the same port. The hardware used to generate analog audio on the Pi, serving as a DAC, is the same hardware used to clock out the precisely timed data that ws2812 pixels demand. It can't do both things at once.

You would need to use a USB audio device, or a Pi-compatible DAC to add an isolated audio output.

UrielGuy, 2016-07-30:
I've tried that, but it still doesn't work, even with an external usb audio card. I might just stick an arduino to do this.

jgarff, 2016-07-31:
This is already on the Wiki in the FAQ.  You'll need to blacklist the Broadcom audio module to keep it from interfering, this will also make it much easier to use USB audio.  I and others have several projects that use USB audio successfully with this library.

closed 2016-07-31 by jgarff