jgarff/rpi_ws281x #106: bad colors with Rpi2

SylvainGirod, 2016-07-12:
Hi,

I'm trying to use NeoPixel Jewels with a raspberry pi 2 and raspbian.
When I run the tests in the python/examples folder (strandtest.py), I don't get the colors that I should.

instead of red, I've got several colors, as blue, green, magenta, yellow etc)
instead of green, I've got a simmilar output with darker colors
instead of blue, nothing displays on the leds.

Do someone got that issue ?

penfold42, 2016-07-13:
Which model do you have ?

It sounds like you have rgbw not RGB LEDs.

Check out the sk6812 branch. It adds support for rgbw devices

SylvainGirod, 2016-07-13:
Hi,

I use the RGB model with integrated drivers (reference adafruit : 2226), so, it must be RGB leds.

Is that possible that the differences between the code and the output is because I have an IMU wired to my pi too ?

penfold42, 2016-07-13:
What's an IMU ?

SylvainGirod, 2016-07-13:
inertial measurement unit. I use it to detect the raspberry pi's position in the space (like a mobile phone can do to adjust its screen).
I need to use it to start or stop the leds depending on the Rpi position

Gadgetoid, 2016-07-13:
The pixel order of ws2812 RGB LEDs is GRB. You can see this on Page 5 of the datasheet linked from Adafruit's store page for the NeoPixel Jewel: https://cdn-shop.adafruit.com/datasheets/WS2812.pdf

```
Composition of 24bit data:
G7 G6 G5 G4 G3 G2 G1 G0 R7 R6 R5 R4 R3 R2 R1 R0 B7 B6 B5 B4 B3 B2 B1 B0
```

The Adafruit NeoPixel Python library explicitly sets this, like so:

```c
ws.ws2811_channel_t_strip_type_set(self._channel, ws.WS2811_STRIP_GRB)
```

You probably need a line not dissimilar to the above in your code.

There are, in fact, a number of different pixel orders and types built into this library which you can see here:

https://github.com/jgarff/rpi_ws281x/blob/4f5d727222ef4e48608d43c51b3b932c0840813d/ws2811.h#L45-L64

Jokelo, 2016-07-15:
I had a similar problem [here](https://github.com/marian42/pixelpi/issues/1).

I agree with @Gadgetoid  above. It definitely seems like the the pixel order is incorrectly setup. From what I've seen, I don't believe you can change the strip type within the strandtest.py without doing some major altercations to the initialization statement. I'd recommend changing the strip type within the [neopixels.py](https://github.com/jgarff/rpi_ws281x/blob/4f5d727222ef4e48608d43c51b3b932c0840813d/python/neopixel.py), and see if that works. You might have to use trial and error to find the right strip type.

If that doesn't work you can try using [SK6812_strandtest.py](https://github.com/jgarff/rpi_ws281x/blob/4f5d727222ef4e48608d43c51b3b932c0840813d/python/examples/SK6812_strandtest.py) and change the LED_STRIP (line 19) to the strip type of your LEDs and see if that works. 

Hope this helps.

jgarff, 2016-07-16:
You can simply change the strip type in the call to Adafruit_NeoPixel() in strandtest.py.  All different strip types are supported.  As for the strip type in your setup, there are several variants, so I believe the problem is simply that yours is different from the default.  I've got several different types of strips, and I often have to use non-default settings.

closed 2016-07-16 by jgarff