jgarff/rpi_ws281x #112: Simple questions..

GodShingShing, 2016-08-10:
First of all, thank you for good libraries.
I'm working with RPi 3 and LEDs (https://www.adafruit.com/products/1734)
I used C codes and python codes.  It works, random flickering. 
But I want to control LED color that I want by using these codes. 
I tried to control LED by using basic functions like 'setPixelcolor'. But it didn't work what I want.
What should I do more? 
Thanks.

sschroe, 2016-08-11:
If the LED is powered with 5V then you need a level shifter to convert the RPIs 3.3V GPIO signal to 5V. Alternatively you can power the LED with just 3.3-4V and it should work without any extras. This is a common source for random flickering.

GodShingShing, 2016-08-12:
Thank you for your reply. 
I powered the LED with 3.3V from RPi 3(pin# 1)  directly. Can't I control LED color by using the codes? Or if you know different code that can control LED color, could you tell me about it? 
Thanks.

sschroe, 2016-08-12:
The first thing you should try is using the example included with this library to see if your LED works with that. If it does not please have a look at issue #103 and try the solution posted there (specifically blacklisting the broadcom audio module). If that still doesn't help the problem is probably with the hardware setup.

closed 2018-03-06 by Gadgetoid