jgarff/rpi_ws281x #115: SWIG dependency

jgmage, 2016-09-14:
I'm using a RPI 3 running Ubuntu Mate 16.04 and installed the library.  Everything built fine and lowlevel.py ran fine but strandtest.py and others generated an error.  The error was:

  File "strandtest.py", line 8, in <module>
    from neopixel import *
  File "build/bdist.linux-armv7l/egg/neopixel.py", line 50, in <module>
  File "build/bdist.linux-armv7l/egg/neopixel.py", line 52, in Adafruit_NeoPixel
AttributeError: 'module' object has no attribute 'WS2811_STRIP_RGB'

After tracking through the code, etc. I was looking into the items imported referencing SWIG.  It turns out that when I installed SWIG from apt, it installed SWIG and SWIG 3.0.  I was able to resolve my problem by installing SWIG 2.0.

If you are having this error, I suggest running:
apt-get install swig2.0

If others can reproduce this, not sure what changes happen with swig3.0 or if the readme may need to be updated.

sschroe, 2016-10-16:
What commands did you use to build? With swig 3.0.10 on archlinux (RPI B+) i used the following commands and it seems to work alright. Though i did have to add some symlinks into the examples folder for the generated files, but i suspect thats normal.

$ swig -python rpi_ws281x.i
$ python setup.py build_ext --inplace

jgmage, 2016-10-17:
Unfortunately I have blown out that OS and changed to Raspian so I only have my repos to refer to but I script the entire RPI build.

Essentially it is:
git clone your repo
sudo scons (in root of cloned git repo - $srcpath)
sudo python $srcpath/python/setup.py install

I didn't pass any command line options (other than install) or create any symlinks.

When I was installing swig from apt I was doing simply apt-get install swig which installed both swig and swig3.0.  After changing this to apt-get install swig2.0 it resolved the issue.

Let me know if you need anything else.

sschroe, 2016-10-17:
Tried it like that and strandtest.py worked on my system, well apart from crashing after a while that is. But before that it did light leds for a while so thats probably unrelated.

```
$ sudo python strandtest.py
Press Ctrl-C to quit.
Traceback (most recent call last):
  File "strandtest.py", line 97, in <module>
    rainbowCycle(strip)
  File "strandtest.py", line 62, in rainbowCycle
    strip.setPixelColor(i, wheel(((i * 256 / strip.numPixels()) + j) & 255))
TypeError: unsupported operand type(s) for &: 'float' and 'int'
```

markusk, 2018-04-02:
+1

I can confirm the first issue on Ubuntu Mate as well. It works with swig2.0 but not with 3.0.

Gadgetoid, 2018-08-29:
^ Note, the above is the correct way to install Python bindings for rpi_ws281x. Don't use the old files in this repository.

See: https://github.com/jgarff/rpi_ws281x/issues/267 and https://github.com/jgarff/rpi_ws281x/pull/317

closed 2018-08-29 by Gadgetoid