jgarff/rpi_ws281x #122: WS2812b wrong sorted colors

SquallHalle, 2016-10-17:
WS2812b:
strip.setPixelColorRGB(1,255,0,0) -> green
strip.setPixelColorRGB(1,0,255,0) -> red
strip.setPixelColorRGB(1,0,0,255) -> blue

penfold42, 2016-10-17:
Did you set the colour order ?

SquallHalle, 2016-10-17:
after this i rewrote order in the python file

penfold42, 2016-10-21:
Look at strandtest_sk6812 and main.c

They both set the strip type which is how the colour order is set.

The order isn't wrong, it's just different for your strip. The ws2811 _chip_ had no fixed colour order - it was based on how you wired it up.
The ws2812 integrated the same chip die with 3 leds in a single package so has a fixed order

Gadgetoid, 2019-11-06:
Assuming fixed and closing because this is 3 years old.

closed 2019-11-06 by Gadgetoid