jgarff/rpi_ws281x #123: Support for using the RPi PCM hardware instead of the PWM hardware.

tvoverbeek, 2016-10-21:
Based on the version of rpi-ws281x used by Pimoroni's UnicornHat (https://github.com/pimoroni/unicorn-hat/tree/master/library/rpi-ws281x) I have made a version which uses the PCM hardware (https://github.com/tvoverbeek/rpi-pcm-ws281x).
I plan to merge this with the PWM version and, when ready, to submit a pull-request so this library (https://github.com/jgarff/rpi_ws281x) supports both PWM and PCM.
Comments?

penfold42, 2016-10-21:
More options are always good.

Do you want to add SPI support as a third option ? https://github.com/penfold42/stuff/blob/master/ws2812_spi.c

tvoverbeek, 2016-10-22:
@penfold42 No plans at the moment. I'll first want to integrate my PCM code. I looked quickly at your ws2812_spi code, and it should be doable. However no promises yet.

tvoverbeek, 2016-11-24:
Pull request submitted. See https://github.com/jgarff/rpi_ws281x/pull/135.
Includes both PCM and SPI support.
SPI support inspired by @penfold42 implementation.

tvoverbeek, 2016-12-11:
Pull request now in dev branch. Please test.

closed 2016-12-11 by tvoverbeek