jgarff/rpi_ws281x #125: Usage Double PWM

hehaaa, 2016-11-11:
To start i really love using this library!
I am currently using the library to control my neo pixel led strips with a raspberry pi 3B.

I'd like to run two separate led steps with the raspberry both playing a different pattern.
I have them connected to pin 18 and 13 for PWM and run my program twice to control both.
This does work however i get some flickering on the leds and sometimes the pattern are switched.
(leds strip 1 playing pattern from led strip 2 and vice versa)

This does not happend when i run a single instance from my program.
Is there any solution for this or am i doing something wrong?

penfold42, 2016-11-12:
I've not tried both at once but from memory a single instance should be using both channels at the same time. The pwm supports stereo audio normally so this is what allows 2 led channels

jgarff, 2016-11-12:
Using both channels is supported, but they must be done from inside the same thread.  The hardware for both channels requires mixing the signals going to the chip logic.  That's why it must be done in the same library instance.

closed 2016-11-12 by jgarff