jgarff/rpi_ws281x #127: WS2812B flickering

oxivanisher, 2016-11-15:
When i run the test ```rpi_ws281x/python/examples/strandtest.py``` (also when i run other things based on this library) I get "flickering". This is a video of my problem: https://youtu.be/UHxVS8SkXOU

- I tried changing from a RPi A to a B+ but i get the exact same behavior.
- I also made sure, that the audio driver is not loaded. The CPU usage is <10% while running the scripts.
- To be sure that it is not because of bad LEDs, I tested them with a Arduino, where they run without this issue.
- The power cabling was the same for the Arduino and the RPi's; also I verified common ground multiple times.
- I switched to diffrent DMA channels without it getting better.

I have no idea how to tackle this problem further.

sschroe, 2016-11-15:
Are you using a level shifter? The refered wordclock project seems to directly connect the GPIO PIN to the LED data. As mentioned under **Hardware:** in the README some extra work is necessary when running the LEDs at 5V.

penfold42, 2016-11-15:
Do you have a separate 5v supply for the leds ?

If you drop the supply voltage to the led strip to about 4v you can quickly achieve the same result as using a level shifter

oxivanisher, 2016-11-15:
This is surely my problem... Would this be a valid level shifter to achieve this?
https://www.sparkfun.com/products/12009

penfold42, 2016-11-15:
I'm not a fan of mosfet level shifters.

I much prefer just using a 74HCT125 (or similar).

Just about any 74HCT series IC will do as long as it has an output pin that can be made to follow an input pin.

74HC isn't guaranteed to work - 74HCT will

oxivanisher, 2016-11-15:
Very good. I just ordered some 74HCT125's and should have them by the end of the week. I will also make corresponding PR's in the wordclock project so the next guy won't make the same error.

Do you know of the top of your head how to connect it up? I will figure it out myself for sure (sooner or later :D )

In any case, thank you very much!

closed 2016-11-15 by oxivanisher

penfold42, 2016-11-15:
There's some discussion here :
https://hyperion-project.org/threads/use-spi-chip-select.279/page-2#post-2666

electrokean, 2016-11-16:
Just an additional recommendation - you should ensure all unused input pins on the 74HCT125 chip are connected to something, like GND (directly or via a 1k resistor).
74HCT chips are based on CMOS technology, and can draw excessive current and fail if an input is left floating - see http://www.nxp.com/documents/user_manual/HCT_USER_GUIDE.pdf for reference

oxivanisher, 2016-11-17:
Would it require one 1k resistor per unused channel (3 pcs) or one to ground and then to the 3 remaining inputs?

penfold42, 2016-11-17:
I would just connect the unused /oe and A inputs to ground directly

electrokean, 2016-11-18:
The resistors aren't really necessary with CMOS, so you can use just one, or just directly connect to GND.
The use of the input resistors goes back to older TTL technology, but many people still find it easier to understand/connect.  You also often don't actually need to remove them if you decide later to use that input for something.

oxivanisher, 2016-11-18:
It works like a charm (the soldering costed me some nerves, but I managed to do it 👍 )!. Feel free to use this high end drawing as you see fit. 😁  
[image]

electrokean, 2016-11-18:
Pins 4, 10, and 13 are also inputs (output enables) and should be tied to GND as well.
Pins 6, 8, and 11 are unused outputs and can be left unconnected.

FocusInfinity, 2016-11-28:
FYI: This problem can be circumvented by installing older RPI noobs OS. I can recommend v1.9.2 published on May 31 in 2016. --> http://downloads.raspberrypi.org/NOOBS/images/

electrokean, 2016-11-29:
@FocusInfinity Actually no, you just can't get around the voltage level issue with a software change.
A 3.3V logic signal as output by the RPi does not meet the logic level requirements of a 5V device like the WorldSemi LEDs.
It may work for you today, but (on another day | with another LED strip | with longer wiring) it may (misbehave | fail completely) due to (variation in manufacturing process | ambient temperature | moon phase | whatever)...

penfold42, 2016-11-29:
Or you have an different cause of the flickering.

Different OS distributions and versions might set up the hardware differently and/or you had onboard audio enabled

bhunziker, 2019-01-31:
For those who would like to try an easy solution before soldering tiny components a simple pull up to 5V helped in my case. I used 1k just on the first LED of as 220 LED long strip and it works perfectly.

kimthostrup, 2019-11-09:
@bhunziker, can you please share a bit more info on how this is accomplished? 
Thanks