jgarff/rpi_ws281x #131: Arduino Simple LED functionality vs. Python

Brobes05, 2016-11-22:
Hello, 

Thank you for this library. I am trying to get my LED strip working with RPI_ws281x; however, only 1/3 of the strand lights up. Incidentally, when I test on Arduino's Simple LED library, everything works fine. Thus, I know it is not a hardware issue. 

I have installed everything properly (as far as I know) since I do not get any errors. I have tried with both Python 2 and 3 and am using the latest version of RPi. Any suggestions would be very appreciated. 
Thank you.

sschroe, 2016-11-22:
Could use some further information on your setup. How are things wired up and what code changes did you make? Are you using a level shifter? Regarding the level shifter you can find more info in #127 if you do not happen to have one.

Brobes05, 2016-11-22:
We are using a separate 5v 10 amp power source and are using pin 12 or pin 18 as a data pin. We are not using a level shifter. Do we need to ground the RPi? It is strange that it works so well with Arduino.

penfold42, 2016-11-22:
The led power supply and rpi must share a common ground.

The ws2812 leds require a data signal voltage of > 3.5v to signal a logical high.
An arduino running at 5v satisfys this. A Pi uses 3.3v signals

Brobes05, 2016-11-22:
Interesting. Thanks. 
In this case. What is the correct way to allow the Pi to output 5v on a GPIO pin while using a 5v adapter? 
If the led power supply and the RPi share a common ground will this be resolved?
Thanks so much.

penfold42, 2016-11-22:
No, a common ground is required and it won't fix your problem with the levels.

Read #127 for info on using a level shifter to shift the 3.3v output to 5v

Brobes05, 2016-11-22:
Thanks. I just saw that. I need tog get a level shifter. Thank you again for all of the help.

Brobes05, 2016-11-27:
Hi, 
I am attempting to level shift PI's 3.3 GPIO to 5V to power my NeoPixel strip. I have a different level shifter (similar to adafruit's 4 channel bi-directional logic level converter -BSS138) than the one in this tutorial: https://learn.adafruit.com/neopixels-on ... -pi/wiring

As a result, I have gotten myself confused in how to correctly wire with my breadboard in regards to the GPIO pin and its relationship to the level shifter and the neoPixel. Any help would be greatly appreciated. I have attached an image with the current state of my wiring. 
[image]

penfold42, 2016-11-27:
The power supply wires have been crossed.

I wouldn't run the 5v for the leds via the breadboard - the current might be too high

The link to the adafrut site is broken

penfold42, 2016-11-27:
According to http://hackaday.com/2008/06/19/sparkfuns-logic-level-converter/
I think the rest of the wiring is ok.

You might also need a stronger pull-up resistor on the output pin to 5v

electrokean, 2016-11-27:
As @penfold42 mentions, the only obvious error in the wiring is the crossed lines from the PSU.  But breadboards are also generally unsuitable for higher current connections (more than a few hundred milli-amps) and that could also be part of the problem.
The bi-directional MOSFET level converters can be a bit slow, leading to signal integrity issues.  Adding lower value pull up resistor(s) can help speed up rise times, but it better to use a single directional 74HCT buffer as discussed in #127

Brobes05, 2016-11-27:
I appreciate the replies and advice. Here is the link that was broken: https://learn.adafruit.com/neopixels-on-raspberry-pi/wiring

I think I understand about the bi-directionality now. I have ordered the chip described in #127; however, I wonder in regards to this MOSFET chip's wiring, if there is a solution to getting around powering the breadboard with 5v. Is this the 1k resistor on the output pin to 5v?

penfold42, 2016-11-27:
Make sure there are 2 5v wires:
1- from power supply to led strip
2- from power supply to level shifter/breadboard.

Yes, I added a 1k resistor between the output of the level shift and 5v.

electrokean, 2016-11-28:
Yes, you don't need bi-directional level shifter - but you want a fast level shifter.
Using the MOSFET level shifter with an extra pullup resistor on the 5V side may help, but the one-directional 74HCT125 buffer is much better, as it is way faster, operates at 5V, and can properly handle the 3.3V logic levels from the RPi at its input.

Brobes05, 2016-11-29:
Thanks again for all of the help. I have received the 74HCT125 chip and it works like a charm. Very glad to have reached this point. 

Very helpful forum and great code!

penfold42, 2016-11-29:
Excellent !

closed 2017-09-16 by jgarff