jgarff/rpi_ws281x #133: User space access to GPIO to run rootless

ymollard, 2016-11-22:
The raspberry Pi team released a user-space API to access the GPIO: https://github.com/raspberrypi/linux/pull/1112

Using `/dev/gpiomem` would allow to run the program rootless.
Can this feature be easily implemented?

sschroe, 2016-11-22:
From what i can tell this does not provide access to the PWM memory areas. Since we need those it is not really usable for this lib.

penfold42, 2016-11-22:
If non root is a requirement there are a couple of options:
1) write a daemon running as root that listens on a named pipe and uses this code for the led interface. Piblaster does this and then you're non root consumer writes to the pipe.
2) use the SPI interface rather than this library. I added this option to Hyperion.

ymollard, 2017-01-03:
@penfold42 I followed your advice to use SPI w/o root, but needed a Python API so I inspired myself from an existing WS2812B + SPI lib. However I experience flashes (the full strip briefly flashing with random colors, approx. every minute but not regular). I noticed than changing the [speed in Hz](https://github.com/doceme/py-spidev#methods) was influencing the quantity of flashes, however I could never fully get rid of them. Maybe changing the delay between blocks could also help. Not sure yet if the problem is the problem is SPI related or WS2812B-protocol related but if you have any clue, feel free :) [I also poked the author](https://github.com/joosteto/ws2812-spi/issues/1).

closed 2017-01-06 by ymollard