jgarff/rpi_ws281x #137: Adding other libraries to scons file (wiringPi)

balettepap, 2016-12-02:
Hello Everyone,

I've been able to use your C library to drive neopixel with a rpi3.
My problem is that I need to use wiringPi library too, but after reading and testing to Scons datasheet, I can't manage to find how to add wiringPi library.

I normally add a -lwiringPi flag to my gcc command, but, even if I add dependancies into SConstruct file, it can't access wiringPi library:
Please see below SConstruct file.
```
`import os

opts = Variables()
opts.Add(BoolVariable('V',
                      'Verbose build',
                      False))

platforms = [ 
    [
        'userspace',            # Target Name
        [ 'linux', 'version' ], # Scons tool (linux, avr, etc.)
        {                       # Special environment setup
            'CPPPATH' : ['/usr/local/include'
            ],
            'LINKFLAGS' : ['/usr/local/lib'
            ],
        },
    ], 
]

#d = env.ParseFlags("-I/usr/local/include -L/usr/local/lib -lwiringPi")
clean_envs = {}
for platform, tool, flags in platforms:
    env = Environment(
        options = opts,
        tools = tool,
        toolpath = ['.'],
        ENV = {'PATH' : os.environ['PATH']},
        LIBS = ['wiringPi'],
    )
    #d = env.ParseFlags("-I/usr/local/include -L/usr/local/lib -lwiringPi")
    env.MergeFlags(flags)
    clean_envs[platform] = env

Help(opts.GenerateHelpText(clean_envs))

Export(['clean_envs'])
SConscript('SConscript');`
```

Any idea ? Thank you very much for your time.

Gadgetoid, 2017-02-15:
I'm probably too late to be of any use, but can you not pre-build the ws281x library and then link against it when compiling your own code, using a regular Makefile or your preferred tooling? It might be easier than grappling with scons.

One of the first things I did when starting to work with this library was hack together a regular Makefile: https://github.com/pimoroni/rpi_ws281x-python/blob/master/library/Makefile

balettepap, 2017-02-15:
> rpi_ws281x-python/blob/master/library/Makefile

Hello,

Yes indeed, but thank you for answering me anyway, it's very nice :)
I finally used python script instead of a C program.

Thanks again, and have a good day !

Hugo

closed 2017-09-16 by jgarff