jgarff/rpi_ws281x #141: interference caused by WiringPi GPIO utility PWM mode

s8mlu, 2016-12-11:
The LEDs works once on a clean reboot of Raspberry Pi. 

But once I use the popular [WiringPi GPIO utility](http://wiringpi.com/the-gpio-utility/) to set Pin 18 to PWM mode, like so:
  `gpio -g mode 18 pwm`
only the first LED works reliably in the strip. The other LEDs will show random colors, or none at all. 

This continues to happen even after I reset the pin back to the default INPUT mode:
  `gpio -g mode 18 in`
The only way that I can find to restore working condition is to reboot. 

I suspect the WiringPi utility has changed some PWM setting which this library depends on. Any ideas?

spAnser, 2016-12-12:
I had this issue when setting up a cronjob under @reboot . If you are trying to do this look into delaying the auto start script I used the answer from [here](http://unix.stackexchange.com/questions/57852/crontab-job-start-1-min-after-reboot) but only used a 20 second delay appeared to work fine for me.

If you are having this issue in general use then. I am not sure what can be done to fix it since I did not have this issue outside of the @reboot auto starting script.

s8mlu, 2016-12-12:
Sorry, nothing to do with @reboot in my case. Once I used the gpio command it would stop working, until I reboot.

spAnser, 2016-12-13:
OK No idea then figured I would post a fix I found out just this past week that worked for me when I experienced your issue only when doing something on @reboot.

s8mlu, 2016-12-13:
Appreciate the tip anyways. I may end up reading the wiringPi source to see what it set in the PWM mode that broke this.

Gadgetoid, 2018-03-06:
Insofar as I can tell, this is an issue with WiringPi (a potentially old and now possibly fixed one?) and not rpi_ws281x? If I'm incorrect, please re-open.

closed 2018-03-06 by Gadgetoid